cyberbotics/webots #101: ROS examples do not run on Windows

gstorer, 2018-12-21:
The following .dll files are missing from the windows installation for 2019a and prevent the ROS examples running on Windows.

libboost_signals-mt.dll
libboost_filesystem-mt.dll

In the mean time while this bug is fixed, can these files be uploaded somewhere so I can run the examples?

DavidMansolino, 2018-12-21:
Hi,

Thank you for the bug report, we are going to fix this issue and send you the missing files as soon as possible.

Best,
David

DavidMansolino, 2018-12-21:
Simply adding the two missing libraries doesn't fix the problem, the controller crashes. It seems the ros libraries and controller has to be re-compiled with the new version of boost.

DavidMansolino, 2019-01-07:
I have tested to recompile the ROS libraries on Windows with the newer version of boost, but this is not a trivial task as this new version of boost doesn't seem compatible with this old version of ROS, we could try to update to the latest version of the ROS libraries but this is not trivial too on Windows.
The ideal solution will be to switch the controller to Python and use rospy which is a pure Python client (http://wiki.ros.org/rospy) this will fix all the compatibility issues and rewriting the controller in Python will allow us to clean and simplify it.
Unfortunately, we won't have the time to do this right now but this is already planned on our medium-term roadmap.

omichel, 2019-02-21:
FYI, we just opened a new PR in #230 that allows you to run a ROS node in a Python Webots controller. This works on Windows, Linux and macOS because it is a pure Python implementation based on rospy.

DavidMansolino, 2019-04-04:
Fixed in #230

closed 2019-04-04 by DavidMansolino