cyberbotics/webots #105: Recompile error for python

wbadry, 2018-12-24:
**Describe the Bug**
I followed the steps described [here](https://cyberbotics.com/doc/guide/using-python#windows-installation). MSYS2 has all requirements. However I get the following error after issuing `make` command
```
swig -c++ -python -outdir "C:/Program\ Files/Webots/lib/python36/" -I"C:/Program\ Files/Webots/include/controller/cpp" -I"C:/Program\ Files/Webots/include/controller/c" -o controller36.cpp controller.i
controller.i:115: Error: Unable to find 'webots/Device.hpp'
controller.i:121: Error: Unable to find 'webots/Accelerometer.hpp'
controller.i:127: Error: Unable to find 'webots/Brake.hpp'
controller.i:154: Error: Unable to find 'webots/camera_recognition_object.h'
controller.i:299: Error: Unable to find 'webots/Camera.hpp'
controller.i:307: Error: Unable to find 'webots/Compass.hpp'
controller.i:313: Error: Unable to find 'webots/Connector.hpp'
controller.i:367: Error: Unable to find 'webots/ImageRef.hpp'
controller.i:368: Error: Unable to find 'webots/Display.hpp'
controller.i:374: Error: Unable to find 'webots/DistanceSensor.hpp'
controller.i:385: Error: Unable to find 'webots/Emitter.hpp'
controller.i:396: Error: Unable to find 'webots/Field.hpp'
controller.i:402: Error: Unable to find 'webots/GPS.hpp'
controller.i:408: Error: Unable to find 'webots/Gyro.hpp'
controller.i:414: Error: Unable to find 'webots/InertialUnit.hpp'
controller.i:420: Error: Unable to find 'webots/Joystick.hpp'
controller.i:426: Error: Unable to find 'webots/Keyboard.hpp'
controller.i:432: Error: Unable to find 'webots/LED.hpp'
controller.i:440: Error: Unable to find 'webots/lidar_point.h'
controller.i:497: Error: Unable to find 'webots/Lidar.hpp'
controller.i:505: Error: Unable to find 'webots/LightSensor.hpp'
controller.i:511: Error: Unable to find 'webots/utils/Motion.hpp'
controller.i:517: Error: Unable to find 'webots/Motor.hpp'
controller.i:542: Error: Unable to find 'webots/mouse_state.h'
controller.i:544: Error: Unable to find 'webots/Mouse.hpp'
controller.i:569: Error: Unable to find 'webots/Node.hpp'
controller.i:575: Error: Unable to find 'webots/Pen.hpp'
controller.i:581: Error: Unable to find 'webots/PositionSensor.hpp'
controller.i:606: Error: Unable to find 'webots/radar_target.h'
controller.i:625: Error: Unable to find 'webots/Radar.hpp'
controller.i:686: Error: Unable to find 'webots/RangeFinder.hpp'
controller.i:698: Error: Unable to find 'webots/Receiver.hpp'
controller.i:706: Error: Unable to find 'webots/Skin.hpp'
controller.i:712: Error: Unable to find 'webots/Speaker.hpp'
controller.i:718: Error: Unable to find 'webots/TouchSensor.hpp'
controller.i:1009: Error: Unable to find 'webots/Robot.hpp'
controller.i:1015: Error: Unable to find 'webots/DifferentialWheels.hpp'
controller.i:1021: Error: Unable to find 'webots/Supervisor.hpp'
make: *** [Makefile:99: controller36.cpp] Error 1
```
All help is appreciated. Also do I have to recompile to use 3.6.5 instead of 3.7 version of python?

omichel, 2019-01-04:
From what I see, it seems that your `WEBOTS_HOME` environment variable was not set properly. The `swig` command line should read:
```
swig -c++ -python -outdir "C:/Program Files/Webots/lib/python36/" -I"C:/Program Files/Webots/include/controller/cpp" -I"C:/Program Files/Webots/include/controller/c" -o controller36.cpp controller.i
```
instead of:
```
swig -c++ -python -outdir "C:/Program\ Files/Webots/lib/python36/" -I"C:/Program\ Files/Webots/include/controller/cpp" -I"C:/Program\ Files/Webots/include/controller/c" -o controller36.cpp controller.i
```
(note the extra backslash `\` in `Program Files`)

Please let me know if that helps.

omichel, 2019-01-04:
And yes, you need to recompile in order to be able to use Python 3.6 instead of 3.7.

omichel, 2019-01-07:
I believe there was indeed a problem in this Makefile. Could you try the fixed version on #108 and confirm that it fixes the problem for you?

wbadry, 2019-01-07:
Thanks @omichel for doing that fix,
Apparently, there several bugs from doing the compilation on Windows.

```powershell
swig -c++ -python -outdir C:/Program\ Files/Webots/lib/python36/ -IC:/Program\ Files/Webots/include/controller/cpp -IC:/Program\ Files/Webots/include/controller/c -o controller36.cpp controller.i
g++ -c -O -Wall -DMS_WIN64 -D_hypot=hypot -Wno-stringop-truncation -IC:/Program\ Files/Webots/include/controller/cpp -IC:/Program\ Files/Webots/include/controller/c -I"/c/Python36/include" controller36.cpp -o controller36.o
In file included from /c/Python36/include/Python.h:53:0,
                 from controller36.cpp:176:
/c/Python36/include/pyport.h:686:2: error: #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
 #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
  ^~~~~
In file included from /c/Python36/include/Python.h:136:0,
                 from controller36.cpp:176:
/c/Python36/include/fileutils.h:41:5: error: '__int64' does not name a type; did you mean '__int64_t'?
     __int64 st_size;
     ^~~~~~~
     __int64_t
In file included from /usr/include/sys/stat.h:22:0,
                 from /c/Python36/include/pyport.h:227,
                 from /c/Python36/include/Python.h:53,
                 from controller36.cpp:176:
/c/Python36/include/fileutils.h:42:12: error: expected ';' at end of member declaration
     time_t st_atime;
            ^
/c/Python36/include/fileutils.h:42:12: error: expected unqualified-id before '.' token
     time_t st_atime;
            ^
/c/Python36/include/fileutils.h:44:12: error: expected ';' at end of member declaration
     time_t st_mtime;
            ^
/c/Python36/include/fileutils.h:44:12: error: expected unqualified-id before '.' token
     time_t st_mtime;
            ^
/c/Python36/include/fileutils.h:46:12: error: expected ';' at end of member declaration
     time_t st_ctime;
            ^
/c/Python36/include/fileutils.h:46:12: error: expected unqualified-id before '.' token
     time_t st_ctime;
            ^
cc1plus: warning: unrecognized command line option '-Wno-stringop-truncation'
make: *** [Makefile:111: controller36.o] Error 1
```

[image]

Moreover, MSYS2 doesn't have any repository for 3.6 and I am totally ignorant of how to make it aware of the version I have on Windows. PowerShell seems solving this but apparently compilation parameters still hold values that are inadequate to windows.

omichel, 2019-01-09:
I just recompiled with Python 3.6.5 and it worked fine for me.
Could you follow the instructions here: https://github.com/omichel/webots/wiki/Windows-installation including the Python optional dependencies and use Python 3.6.5 instead of 3.7.x.
In your ~/.bash_profile file, set `export PYTHON37_HOME=/D/Python36` to use Python 3.6 instead of 3.7.
And recompile the whole webots, including the Python wrapper.
Let me know if that works.

omichel, 2019-01-09:
*Note*: I installed Python 3.6.5 from https://www.python.org/downloads/release/python-365/ and more specifically https://www.python.org/ftp/python/3.6.5/python-3.6.5-amd64.exe

closed 2019-01-09 by omichel

wbadry, 2019-01-23:
Hi @omichel ,
Did you recompile it using MSYS2? Python is not detected on MSYS2 if installed on Windows.

omichel, 2019-01-23:
Yes, we use MSYS2 to recompile everything.

Normally, if Python is in your PATH environment variable, you should be able to execute it from MSYS2.
It should be in your PATH environment variable if you followed the installation instructions for Python at https://github.com/omichel/webots/wiki/Windows-optional-dependencies#python including the modification of your `~/.bash_profile` file with `PYTHON37_HOME` pointing to your Python 3.6 installation folder (you need to restart MSYS2 after that).

Can you type `printenv PATH` in your MSYS2 console and check if you see the path to your Python 3.6 folder? You can also check `printenv PYTHON37_HOME`.

**Beware**: you should avoid any spaces in your paths, e.g., your `Webots Projects` folder should probably renamed `Webots_Projects` (use a `_ ` instead of of space ` `) to avoid problems with compilation.

wbadry, 2019-01-23:
Thanks Oliver.
Actually I carefully followed the steps in [windows installation guide](https://github.com/omichel/webots/wiki/Windows-installation) in addition to installation of [optional dependencies](https://github.com/omichel/webots/wiki/Windows-Optional-Dependencies).

1. Python 3.6 is downloaded and already in Windows Environment as tested on **Windows Powershell**

[image]

[image]

[image]

2. Bash_Profile was modified

[image]

3. I renamed **webots folder**

[image]

4. Here is the **print of MSYS2** 

[image]

When I perform `make -j8` as I have 8 core processor, I encounter this error. I think there might be a path problem in either MSYS2 path or webots path as it shows no such file or directory when copying `ode.dll` during compilation process. Also Python is still not recognized in MSYS2 environment. Perhaps this path would be recognized during compilation of the script which is so far stopped at that point in shown in the image.

omichel, 2019-01-23:
It seems your `PATH` environment variable under MSYS2 doesn't contain the path to Python.
Normally, you `~/.bash_profile` should call the `master` alias at the end which calls in turn the `msys.sh` script which adds the path to Python in your `PATH` environment variable. Some some reason, this doesn't seem to be executed. Can you type `master` in your MSYS2 console to attempt to execute it?

It may be worth to reinstall your `~/.bash_profile` file appropriately... Be sure to have it appended to the MSYS2 user `.bash_profile` file which already exists after installing MSYS2, as explained in the installation instructions. This file is executed each time you open the MSYS2 console.

wbadry, 2019-02-10:
I tried but it seems I ended up having same error. May I request a release of webots compiled with Python 3.6? I would really appreciate it. Thanks

omichel, 2019-02-11:
Waleed, I just sent you an e-mail about it.

wbadry, 2019-02-11:
Hi Oliver,
Thank you so much for your email. I appreciate your offer to **build me a 3.6 Python release for 160 USD**. However, I would appreciate leaving this case open for other members that may do it without payment. Thank you so much.

omichel, 2019-02-11:
Up to you.

wbadry, 2019-02-14:
A quick update. tensorflow now supports version Python 3.7 so all good. Thank you @omichel so much for your consistent support. You are so amazing 👍

omichel, 2019-02-14:
Good news! Thanks for the update.

PR #108 (2019-01-09): Fixed used of space protected WEBOTS_HOME_PATH variable
Fixes #105.