cyberbotics/webots #127: Webots crashes on startup

fabienrohrer, 2019-01-22:
**Describe the Bug**

On some hardware, Webots R2019a crashes on startup.

**Screenshots**

[image]

_(from ticket #324)_ 

**System**
 
It seems to affect mainly Windows users, without guarantee for now.

fabienrohrer, 2019-01-22:
A user fails the procedure at step 2. He can run "empty.wbt" but not "geometric_primitives.wbt" whatever the preferences of step 3.

DavidMansolino, 2019-01-22:
It seems the problem comes from HDR background. Non HDR background + PBR + Bloom + GTAO is working fine.

DavidMansolino, 2019-01-23:
The problem seems to be the backing of the specular irradiance, disabling it fixes the crash.

DavidMansolino, 2019-01-23:
Reducing the resolution of the specular irradiance cubemap texture from 512 to 128 (and also the number of mimaps from 10 to 8) fixes the problem, I can now run all the worlds of the guided tour.

Since this resolution is not yet linked to the texture quality preference I propose to link it to it, so that it will work in low-quality texture mode.
Then we will need a way to detect these problematic GPUs and switch automatically to the low texture quality mode (the GPU I am currently testing have 1Gb of dedicated memory which is not so low).

DavidMansolino, 2019-01-23:
I have made a few comparisons to check the impact on the rendering (please make sure to compare them at full resolution and not only the small preview):

| resolution | metalness 1 | metalness 0 |
| ------------- | ---------------- | --------------- |
|     512       | [image] | [image] |
|     256 | [image] | [image] |
|     128  | [image] | [image] |
|      32  |  [image] |  [image] |
|       8 | [image] |  [image] |

As the difference is almost invisible between 512 and 256, here is what I propose:
  - Put 256 instead of 512 by default.
  - Put 128 in medium texture resolution.
  - Put 64 in low texture resolution.
  - Switch automatically to 'medium' texture resolution for GPU with less than 2gb of memory.
  - Switch automatically to 'low' texture resolution for GPU with less than 1gb of memory.

omichel, 2019-01-23:
That seems to be a very good proposal.

PR #133 (2019-01-30): Fix Crash due to Baking Size
Fix #127
closed 2019-01-30 by DavidMansolino