cyberbotics/webots #137: Unable to Load JPEG Textures from cmd.exe

DavidMansolino, 2019-01-25:
**Describe the Bug**
JPEG textures doesn't load and produce the following error message:
```
WARNING: Texture image size of 'C:/Program Files/Webots/projects/objects/obstacles/protos/textures/oil_barrel_bottom_occlusion.jpg' is not a power of two: rescaling it from -1x-1 to 0x0.
WARNING: OilBarrel (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot load texture 'C:/Program Files/Webots/projects/objects/obstacles/protos/textures/oil_barrel_bottom_occlusion.jpg': Unsupported image format.
```

This bug was reported by several users and I was finally able to reproduce it by starting Webots from `cmd.exe`. It is working fine when launched from the desktop shortcut.

**Steps to Reproduce**
1. Launch Webots with `cmd.exe`

**Expected behavior**
The textures should load without any problem.

**Screenshots**
[image]

**System**
 - Operating System: Windows 10 (only from cmd.exe)

omichel, 2019-01-25:
I could reproduce the problem as well, but I notice that it can be fixed by starting `webots.bat` instead of `webots.exe`. This is probably due to the fact the DOS cmd.exe ignores the `webots.exe.config` file that adds an extra path to the search for shared libraries (e.g., `../../usr/bin`).

omichel, 2019-01-29:
We should ask the users to launch `webots.bat` instead of `webots.exe` to see whether that fixes the problem for them.

PR #222 (2019-02-19): Added instructions to launch Webots from a DOS console
 Fixes #137.

👍
PR #1532 (2020-04-17): Renamed webots.exe to webots-bin.exe
Fixes #137 as it was reported again by several users.
This fix simply renames `webots.exe` to `webots-bin.exe`.
It fixes the problem of typing `webots` in a DOS CMD console.
Previously, this was launching `webots.exe` directly and was causing problems with JPEG textures as explained in #137.
Now, it launches a different `webots.exe` which is a simple launcher for `webots-bin.exe`.
`webots.exe` sets the `PATH` environment variable appropriately and starts `webots-bin.exe`.
As a consequence, Webots launches nicely and the JPEG textures bug doesn't show up any [... cut]
closed 2019-02-19 by omichel