cyberbotics/webots #147: Check for updates... TLS initializaton failed

aiaustin, 2019-01-31:
On R2019a on Windows 10 Pro... when using menu... Tools -> Check for updates... pop up message indicates...

Cannot get the Webots current version due to: "TLS initializaton failed"

omichel, 2019-01-31:
Thank for reporting this bug. I could reproduce it. We are going to try to fix it as soon as possible. Meanwhile, you will have to check manually for updates. FYI, we plan to release a new version next week or so that will hopefully fix this problem.

omichel, 2019-01-31:
I believe I have found and fixed the problem. Can you please download this [libssl-1_1-x64.zip](https://github.com/omichel/webots/files/2816865/libssl-1_1-x64.zip) ZIP file, unzip it and move it to `C:\Program Files\Webots\msys64\mingw64\bin\libssl-1_1-x64.dll`. Then restart Webots and let me know if that fixes the problem.

aiaustin, 2019-01-31:
Yes Olivier, that's fixes the problem.  Webots R2019a with that DLL in place reports its correct version and that its up to date (as of today). Thanks.

omichel, 2019-01-31:
Thank you for the confirmation. The bug is now fixed in #148 and will be integrated into Webots R2019a-rev1 to be released next week or so.