cyberbotics/webots #150: webots-bin fails to link with OIS library on Linux

khklau, 2019-02-03:
**Description**
I'm getting the following linking error when building webots-bin on both Ubuntu 18.04 and Centos 7:

```
# linking  ../../bin/webots-bin
build/release/WbJoystickInterface.o: In function `WbJoystickInterface::init()':
WbJoystickInterface.cpp:(.text+0x64e): undefined reference to `OIS::InputManager::createInputObject(OIS::Type, bool, std::string const&)'
WbJoystickInterface.cpp:(.text+0xa9f): undefined reference to `OIS::InputManager::createInputSystem(std::multimap<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > >&)'
```

I've carefully followed the build instructions for Linux described in the [Linux installation wiki page](https://github.com/omichel/webots/wiki/Linux-installation) and installed all of the required dependencies plus the optional dependencies for the OpenStreeMap importer.

I noticed [dependencies/Makefile.linux](https://github.com/omichel/webots/blob/master/dependencies/Makefile.linux) is setup to download a pre-built OIS 1.4.0 library. I first suspected that either:
1. the pre-built library had some strange conditional compilation that left out the definitions of `createInputObject` and `createInputSystem` or
2. it was built with a GCC version that has an incompatible ABI

so I removed libOIS from [dependencies/Makefile.linux](https://github.com/omichel/webots/blob/master/dependencies/Makefile.linux), installed the libOIS 1.3.0 library from Ubuntu's repository and changed [src/webots/Makefile](https://github.com/omichel/webots/blob/master/src/webots/Makefile) to link against this libOIS version but I still got the same result, so I'm not sure what's going on.

**Steps to Reproduce**
1. Install the dependencies listed in the [Linux installation wiki page](https://github.com/omichel/webots/wiki/Linux-installation)
2. Installed the dependencies listed in the [OpenStreetMap Importer page](https://www.cyberbotics.com/doc/automobile/openstreetmap-importer#linux)
3. `git submodule init`
4. `git submodule update`
5. `make -j 4`

**System**
1. Ubuntu machine
 - Operating System: Ubuntu 18.04
- CPU Architecture: x86_64
- C++ Compiler: Gcc 7.3 (the default on Ubuntu 18.04)
- Linker: GNU LD 2.30 
2. Centos machine
- Operating System: Centos 7.6.1810
- CPU Architecture: x86_64
- C++ Compuler: Gcc 4.8.5 (the default on Centos 7)
- Linker: GNU LD 2.27-34.base.el7

**Additional context**
For reference here is the linking command being used on my machine:
```
g++ -rdynamic -s -o ../../bin/webots-bin build/release/WbRobotWindowTransportLayer.o ... (long list of object files) .. build/release/WbZoom.moc.o ../../dependencies/lua-5.2.3/src/liblua.a ../wren/libwren.a ../glad/libglad.a -Wl,-rpath ../../lib -L../../lib -lQt5Core -lQt5Network -lQt5Gui -lQt5OpenGL -lQt5WebSockets -lQt5Widgets -lQt5PrintSupport -lQt5WebEngine -lQt5WebEngineCore -lQt5WebEngineWidgets -lQt5Multimedia -lQt5MultimediaWidgets -lQt5Sql -lQt5Sensors -lQt5WebChannel -lQt5Xml -lopenal -lGL -lode -lusb -lGLU -ldl -lpci -lOIS -lcrypto -lpico -lfreetype -lassimp
```

khklau, 2019-02-03:
I forgot to mention that this problem exists on both the R2019a tag and the latest commit on master [5284af1](https://github.com/khklau/webots/commit/5284af169cd47ee34a9adf0a213c04acb09609fb).

DavidMansolino, 2019-02-04:
I just tried on Ubuntu 18.04 (x86_64) with the exact same compiler and liker versions and I can't reproduce the problem.

Can you please check that these files exist in `webots/lib`: `libOIS.so` and `libOIS-1.4.0.so`?
It would also be good to check if running the following commands from `webots` solves the problem: `make cleanse` and then `make` (this will ensure to reinstall the dependencies in case something went wrong the first time).
Finally, you could also try to recompile OIS 1.4 for your machine, we have documented the compilation instructions here: https://github.com/omichel/webots/wiki/OIS-compilation

DavidMansolino, 2019-04-05:
This is not reproducible.

closed 2019-04-05 by DavidMansolino