cyberbotics/webots #171: Improve linear_motor.wbt Node hierarchy

DavidMansolino, 2019-02-07:
**Describe the Bug**
The node hierarchy of the linear motor simulation should be improved. It does contain useless group node and empty transform node.

Furthermore with the AMD radeon R5 the ruler is to much transparent.
Nvidia GTX 980:
[image]

Radeon R5:
[image]

fabienrohrer, 2019-02-11:
The issue with the transparency is related with the `OpenGL/textureQuality` parameter. At low, the object is transparent, at high it is less transparent and more blue. I will investigate why.

fabienrohrer, 2019-02-11:
The smooth scale operation of Qt image clearly alternates the alpha value (3 times or so):
https://github.com/omichel/webots/blob/master/src/webots/nodes/WbImageTexture.cpp#L138

PR #197 (2019-02-13): Fix texture downscaling (alternative 2)
Fix #171

Do a Gaussian blue while scaling down.
closed 2019-02-13 by fabienrohrer