cyberbotics/webots #172: X3D/VRML Export Color Missmatch

DavidMansolino, 2019-02-07:
**Describe the Bug**
When both the `baseColor` and `baseColorMap` of the `PbrAppearance` node are set, the export to X3D or VRML is not working as expected.

**Steps to Reproduce**
1. Export the `city.wbt` world to X3D

**Expected behavior**
The color should be very close.

**Screenshots**
Webots:
[image]
X3D:
[image]

**System**
 - Operating System: Windows 10 and Linux Ubuntu 16.04
 - Graphics Card: NVIDIA GeForce GTX 980 and AMD Radeon R5

fabienrohrer, 2019-02-07:
It's because of X3DOM which doesn't support Appearance.diffuseColor + Appearance.texture.
Solutions for this are not obvious, if we drop the Appearance.diffuseColor coachwork will be fixed, if we drop Appearance.texture, the texture will be lost (this can cause issues on textures with alpha).

Generating a texture is time consuming (to implement and to load ^^ ).

This will be solved directly when we will use THREEjs. => I postpone this to R2019b.

DavidMansolino, 2019-02-07:
Maybe exporting only the color in this case as a temporary solution could be good (quickly implemented and working better than the current state).

fabienrohrer, 2019-02-26:
I didn't added `Moose` to `web_component_studio` because of this issue.
When fixed, we need to add the following to `components.json`:

```json
"projects/robots/clearpath/moose/protos/Moose.proto": {
  "background": [1.0, 0.0, 1.0],
  "fields": "translation 0 0.38 0 rotation 0 1 0 5.76"
}
```

DavidMansolino, 2019-04-04:
Linked to #254

stefaniapedrazzi, 2019-04-30:
I confirm this issue is fixed in R2019b by #254.

closed 2019-04-30 by stefaniapedrazzi