cyberbotics/webots #175: Impossible to Recompile Qt Robot-Windows from the Windows Release

DavidMansolino, 2019-02-07:
**Describe the Bug**
Recompilation of Qt robot-windows from the Windows release fails with the following error:
```
# linking robotis-op2_window.dll
build/release/Viewer.o:Viewer.cpp:(.text+0x160): undefined reference to `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/Viewer.o:Viewer.cpp:(.text+0x160): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/Viewer.o:Viewer.cpp:(.text+0x1b0): undefined reference to `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/Viewer.o:Viewer.cpp:(.text+0x1b0): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/Viewer.o:Viewer.cpp:(.text$_ZN5QListI7QStringED2Ev[_ZN5QListI7QStringED2Ev]+0x97): undefined reference to `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/Viewer.o:Viewer.cpp:(.text$_ZN5QListI7QStringED2Ev[_ZN5QListI7QStringED2Ev]+0x97): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0x1f3): undefined reference to `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0x1f3): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0x30d): undefined reference to `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0x30d): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0x573): more undefined references to `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)' follow
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0x573): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0x68d): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0x903): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0xa1d): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0xc93): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `QArrayData::deallocate(QArrayData*, unsigned long, unsigned long)'
build/release/TransferWidget.o:TransferWidget.cpp:(.text+0xdad): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
make: *** [C:/Program Files/Webots/resources/Makefile.include:642: build/release/robotis-op2_window.dll] Error 1
```

**Steps to Reproduce**
1. Open the soccer World.
2. Open the `plugins/robot_windows/robotis-op2_window_Makefile` file in the text editor.
3. Try make clean and then make (you probably need to relocate the project including plugins).

**Expected behavior**
It should be possible to recompile the robot window from Webots without any warnings/errors.

**System**
 -  Operating System: Windows 10 (only from the distribution, working fine from the development environment)

**Additional context**
Probably related to #174

DavidMansolino, 2019-02-07:
Same issue with the `gui_tracker` simulation.

omichel, 2019-02-08:
It seems indeed impossible to link with Qt5 shared libraries when recompiling with gcc7 on MSYS2. I assume this is because Qt5 C++ libraries were compiled with gcc8. So, there is no simple solution to this problem. Since the native robot windows are officially deprecated and provided only for convenience, I would keep it like that for R2019a revision 1 (native robot windows not recompilable outside of the development environment).

omichel, 2019-06-03:
Fixed in #506.

closed 2019-06-03 by omichel