cyberbotics/webots #190: Main 3D View Anti-Aliasing Preference Should be a Checkbox

DavidMansolino, 2019-02-11:
**Describe the Bug**
The main 3d view anti-aliasing preference should be a checkbox since the only options are 'enabled' or 'disabled'.

**Screenshots**
[image]

omichel, 2019-02-18:
This should be cleaned up when we implement camera post-processing (including GTAO). Also, I believe we should remove the distinction between camera and main view. The new preferences should simply offer to enable shadows, GTAO, SMAA and texture quality. This will affect both the main view and robot cameras.

DavidMansolino, 2019-02-19:
> This should be cleaned up when we implement camera post-processing (including GTAO).

This is already implemented on develop.

>  believe we should remove the distinction between camera and main view.

:+1:

PR #225 (2019-02-19): Cleanup-up anti-aliasing preferences
Fixes #190.
closed 2019-02-20 by omichel