cyberbotics/webots #201: Crash wiht high texture quality on load using NVIDIA graphics card

stefaniapedrazzi, 2019-02-13:
**Describe the Bug**
Webots crashes loading worlds if the Texture Quality is set to High.
My NVIDIA graphics card is not detected as low-gpu and by default the texture quality is set to high.

**Steps to Reproduce**
1. Install Webots / reset Webots configurations
2. Go to 'webots_box.wbt'

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce GT 640M with 2GB dedicated VRAM

**Stack**
```
Thread 14 received signal ?, Unknown signal.
[Switching to Thread 7080.0x1b54]
0x000000001eef9389 in vk_optimusGetInstanceProcAddr ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
(gdb) bt
#0  0x000000001eef9389 in vk_optimusGetInstanceProcAddr ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#1  0x000000001eafe7b3 in nvoglv64!DrvValidateVersion ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#2  0x000000001eb4e99f in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#3  0x000000001eb4cd86 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#4  0x000000001eb3ad56 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#5  0x000000001ec4395e in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#6  0x000000001ec46714 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#7  0x000000001ec459ed in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#8  0x000000001ec46c1b in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#9  0x000000001ec5e029 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#10 0x000000001eafb94e in nvoglv64!DrvValidateVersion ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#11 0x000000001ec1fe52 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#12 0x000000001ebe3fc6 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#13 0x000000001ec44327 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#14 0x000000001ec41a47 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#15 0x000000001ec422b1 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#16 0x000000001ec0af37 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#17 0x000000001ec16d44 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#18 0x000000001ec0ac93 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#19 0x000000001ec18dab in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#20 0x000000001ebb3e83 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#21 0x000000001ec13189 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#22 0x000000001ec09382 in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#23 0x000000001ec115bd in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#24 0x000000001ec133dc in nvoglv64!DrvPresentBuffers ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#25 0x000000001e871a43 in ?? ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#26 0x000000001e88042a in ?? ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#27 0x000000001e71d66d in ?? ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#28 0x000000001e8b1e70 in ?? ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#29 0x000000001e8b1c20 in ?? ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#30 0x000000001eb00d6b in nvoglv64!DrvValidateVersion ()
   from C:\WINDOWS\System32\DriverStore\FileRepository\nvaci.inf_amd64_cbe7e13da3856715\nvoglv64.dll
#31 0x00007ffcda653034 in KERNEL32!BaseThreadInitThunk ()
   from C:\WINDOWS\System32\kernel32.dll
#32 0x00007ffcdc783691 in ntdll!RtlUserThreadStart ()
   from C:\WINDOWS\SYSTEM32\ntdll.dll
#33 0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

DavidMansolino, 2019-02-13:
Can you try replacing 'resolution' by '128' (with the texture quality set to high) here:
https://github.com/omichel/webots/blob/master/src/webots/nodes/WbCubemap.cpp#L246

DavidMansolino, 2019-02-13:
2GB is the limit, on AMD it will switch to medium quality (because we get only the available memory), but on NVIDIA since we get the total memory it will stay high, maybe we can just replace the 'smaller than' condition by 'smaller or equal to':
https://github.com/omichel/webots/blob/master/src/webots/gui/WbView3D.cpp#L1241

stefaniapedrazzi, 2019-02-13:
> Can you try replacing 'resolution' by '128' (with the texture quality set to high) here:
> https://github.com/omichel/webots/blob/master/src/webots/nodes/WbCubemap.cpp#L246

With this change it works.
And it also works if I switch the quality to medium.

stefaniapedrazzi, 2019-02-13:
By the way on my Linux Desktop I also have a NVIDIA graphics card with 2 GB of memory (GeForce GT 740) and texture quality set to high works fine.

DavidMansolino, 2019-02-13:
> By the way on my Linux Desktop I also have a NVIDIA graphics card with 2 GB of memory (GeForce GT 740) and texture quality set to high works fine.

The Linux and Windows drivers are probably not the same and they support different maximum texture size.

stefaniapedrazzi, 2019-02-13:
Yes it was just to mention that increasing the memory required to use high quality textures is good but not necessary on all the system.

PR #202 (2019-02-13): Fix crash on NVIDIA 2Gb GPU.
Fix #201
closed 2019-02-13 by DavidMansolino