cyberbotics/webots #206: Telemetry Issue with Bolean Data

DavidMansolino, 2019-02-13:
**Describe the Bug**
All the bolean data recorded by the telemetry are wrong.

**Steps to Reproduce**
1. Disable the camera anti-aliasing.
2. Open any world.
3. Check your entry in https://cyberbotics.com/telemetry => the  camera anti-aliasing is not marked as disabled

**System**
 - Operating System: Windows 10

**Additional context**
Probably an issue with the default value type provided when retrieving the value, e.g.: https://github.com/omichel/webots/blob/master/src/webots/core/WbTelemetry.cpp#L60

DavidMansolino, 2019-02-13:
The problem seems to be on the server side, on Webots side we are sending correct data.

omichel, 2019-02-13:
I just fixed it.

closed 2019-02-13 by omichel

DavidMansolino, 2019-02-13:
Thank you, I confirm it is now working fine!