cyberbotics/webots #211: Streaming failures on Linux

omichel, 2019-02-13:
**Describe the Bug**
The streaming mode either crashes Webots or hangs.

**Steps to Reproduce**
1. webots --stream projects/samples/devices/worlds/motor.wbt
2. open resources/web/streaming_viewer/index.html and connect

&rarr; the loading of the 3D view in the web browser hangs

1. webots --stream projects/vehicles/automobile/highway.wbt
2. open resources/web/streaming_viewer/index.html and connect

&rarr; Webots crashes

**Screenshots**
[image]

**Note:** see the strange `textureTransform="` text above the 3D view.

**System**
 - Operating System: Ubuntu 18.04 running in VMware on Windows 10 with hardware acceleration enabled
 - Graphics Card: NVIDIA GeForce GTX970

fabienrohrer, 2019-02-13:
I cannot reproduce on macOS native. Streams work as expected.

omichel, 2019-02-13:
It may be due to some OpenGL bug on VMware (again).

fabienrohrer, 2019-02-13:
I checked and it's not related with textureQuality (my first hypothesis reading this)

omichel, 2019-02-13:
I just tested on Windows and it works as expected.

omichel, 2019-02-13:
Can someone test it on a native Linux, to see if it's VMware related?

DavidMansolino, 2019-02-13:
>Can someone test it on a native Linux, to see if it's VMware related?

I am doing this right now.

fabienrohrer, 2019-02-13:
On my VM, I have a browser crash (!) on highway.wbt, but 1. motor.wbt works smoothly.

DavidMansolino, 2019-02-13:
On my native Linux both are working (highway is a bit slow in the web browser while running).

omichel, 2019-02-13:
Yes, I would assume this highway.wbt is too heavy (texture wise) for a VM graphics to run both Webots and the webGL view, so let's close this issue as I see no simple way to fix it.

omichel, 2019-02-13:
Thank you for the tests!

closed 2019-02-13 by omichel

fabienrohrer, 2019-02-13:
Ok that sounds good.