cyberbotics/webots #214: Field changes not propagated for nested PROTO slots in Tinkerbots worlds

stefaniapedrazzi, 2019-02-14:
**Describe the Bug**
The robot/3d view is not updated when changing the translation/rotation of the TinkerbotsCubieBoxWithRoundSlots PROTO in the racer.wbt world.

**Steps to Reproduce**
1. Go to racer.wbt
2. Select node on  in the scene tree
3. Change translation
4. See error -> nothing changes in the 3d scene

**Expected behavior**
The position of the TinkerbotsCubieBoxWithRoundSlots object in the 3d scene should change

RobSol93, 2020-07-29:
By deleting the nested PROTO (TinkerbotsCubieBoxWithRoundSlots in this example), then importing it again, the translation/rotation is working again. It also works after saving the world, but when reloading the world (or loading this saved world), it stops working.

RobSol93, 2020-07-31:
The problem seem to be rooted in trying to set the translation/rotation field of first node in the nested PROTO.

The **TinkerbotsCubieBoxWithRoundSlots** PROTO in **ThinkerbotsBase** has **translation/rotation** field  linked by **IS** to its first node, **Slot**. Same goes for its own nested PROTO, **TinkerbotsCubieFemaleCube**, which also doesn't respond to change in the fields.

However, if one look at **TinkerbotsMotor**, the _translation **IS** translation_ is not in the first **Slot** node, but the child **Solid**. And this PROTO can be both translated and rotated. I'm not sure if this is a bug or by design, but knowing this, the issue should be easy to work around.

DavidMansolino, 2020-07-31:
I am not sure to understand the difference between `TinkerbotsCubieBoxWithRoundSlots` and `TinkerbotsMotor`.

`TinkerbotsCubieBoxWithRoundSlots`
```
  Slot {
    type "lego cross+"
    endPoint Solid {
      translation IS translation
```
`TinkerbotsMotor`
```  
Slot {
    type "tinkerbots+"
    endPoint Solid {
      translation IS translation
```
That looks like exactly the same hierarchy no?

stefaniapedrazzi, 2020-08-03:
This is an internal bug in the instantiation of nested PROTO nodes.

There could be many different reasons why it works in some cases and not in others.
And it is also likely that the difference is in the slot where the PROTO is inserted and not only the PROTO part structure.
But the design of all the Tinkerbots PROTO parts is correct and they should all work in any cases.

RobSol93, 2020-08-03:
You are absolutely correct. I was able to move my nested PROTOS by doing what I wrote, and I guess I was too brief when reading the Tinkerbots code.

I discovered today that my nested PROTOS problem was not solved after all. When running the simulation, the absolute position of my PROTOS are not updated. I can move my Solid (child of root node) within my PROTO, but the parent/root node (Transform) of the PROTO does not seem to update.

It is quite weird, because PROTOs that are children of PROTOs move when their parent move, but the absolute position of these children does not update (the absolute position definitely changes)

PR #2063 (2020-08-04): Fix nested PROTO field redirection
Fix #214: internal PROTO nodes are reset flags needed to correctly redirect fields.
closed 2020-08-04 by stefaniapedrazzi