cyberbotics/webots #215: TinkerbotsCubieBoxWithRoundSlots slots

fabienrohrer, 2019-02-14:
`TinkerbotsCubieBoxWithRoundSlots.leftSlot` and `TinkerbotsCubieBoxWithRoundSlots.axisSlot` are located at the same position, and `TinkerbotsCubieBoxWithRoundSlots.front/backSlot` are missing.
We should certainly drop `TinkerbotsCubieBoxWithRoundSlots.leftSlot` and `TinkerbotsCubieBoxWithRoundSlots.rightSlot`.

stefaniapedrazzi, 2019-02-14:
Note that the slots name/position in `TinkerbotsCubieBoxWithRoundSlots` have at least to be consistent with the slots in `TinkerbotsCubieBoxWithCrossSlots` (exact same shape).
This means that the slots in `TinkerbotsCubieBoxWithCrossSlots` have to be updated as well.

fabienrohrer, 2019-02-14:
I also remarked a very minor issue which could be fixed here: the `TinkerbotsCube` slots are not sorted consistently (`backSlot` and `frontSlot` are inverted).

stefaniapedrazzi, 2019-02-14:
Left/right slot positions seems to not be consistent in `TinkerbotsCubieMaleCube` and `TinkerbotsCubieFemaleCube`.

stefaniapedrazzi, 2019-02-18:
> Left/right slot positions seems to not be consistent in `TinkerbotsCubieMaleCube` and `TinkerbotsCubieFemaleCube`.

I checked again and they are correct in Webots.

stefaniapedrazzi, 2019-02-18:
> I also remarked a very minor issue which could be fixed here: the `TinkerbotsCube` slots are not sorted consistently (`backSlot` and `frontSlot` are inverted).

If I understand correctly the issue, i.e. that based on the up, down, left and right faces the position of the back and front faces are inverted, then this is affecting also TinkerbotsDistanceSensor, TinkerbotsLightSensor, and TinkerbotsMotor.

Given that for the TinkerbotsDistanceSensor and TinkerbotsLightSensor it makes sense to have the device itself on the front face, then maybe it is better to switch the right and left faces.

fabienrohrer, 2019-02-18:
Oops, I was speaking on a much simplest issue: the definition of the `back/frontSlot` are strangely sorted here (e.g. `backSlot` appears on top of the `frontXSlots` while `frontSlot` appears on top of the `backXSlots`):
https://github.com/omichel/webots/blob/master/projects/robots/tinkerbots/protos/TinkerbotsCube.proto#L198

What you raise here is indeed another issue :-(

closed 2019-02-20 by stefaniapedrazzi