cyberbotics/webots #240: A Python example for using external editor with webots controller and camera

wbadry, 2019-02-23:
It would be great if python example could be added which uses Python for interacting with any webots robot controller externally. Also if a camera stream could be retrieved as well. Thanks

omichel, 2019-02-25:
Yes, this is possible to implement a controller that acts as a TCP/IP relay to an external Python or whatever program. An example of this is provided in https://github.com/omichel/webots/tree/master/projects/robots/k-team/khepera1/controllers/tcpip where a C client program connect to a TCP/IP relay controller (named tcpip). A Python client could connect the very same way. However, you may need to modify the TCP/IP relay client to support all the sensors and actuators you actually need.

Alternatively, a better and more generic way would consist in implementing a special mode in Webots to launch controller programs, e.g., for example instead of having Robot.controller "my_controller", we could specify Robot.controller "pipe:98327" or "tcp/ip:47366" to refer to a pipe file or a TCP/IP port to which a standalone controller would connect. Webots will then wait for controller to get connected instead of launching the controllers. Of course, the libController should be modified to allow to launch a controller program (in any language, including Python) in a standalone way. I believe the "pipe:xxxx" protocol is easy to implement in Webots source code as Webots currently rely on local pipes to communicate with controllers. The "tcp/ip" protocol is more difficult to implement (as we have to serialize the way camera images are sent. With local pipes, camera images are sent to the controller via a shared memory segment, which will not work over TCP/IP.

@wbadry: If you feel like you could implement one of these options as a contribution to Webots, I would be happy to review your pull request about it. If you would like that someone else implements it for you, please contact sales@cyberbotics.com about it and you will receive an offer for it.

DavidMansolino, 2019-08-14:
Partially fix in #807

The documentation is available here: https://www.cyberbotics.com/doc/guide/using-pycharm-with-webots

PR #807 (2019-08-14): Document PyCharm with Webots
Fix #240.

https://www.cyberbotics.com/doc/guide/using-pycharm-with-webots?version=doc-pycharm
closed 2019-08-14 by DavidMansolino