cyberbotics/webots #268: General/threadingPolicy Preference Usefulness

DavidMansolino, 2019-03-04:
**Describe the Bug**
The ``General/threadingPolicy`` preference is not documented anywhere and it is not possible to change it from the user interface.
Does someone know what it is/was used for? Is it still used (in this case it should be documented, at least on the wiki if used for debugging) or can we remove it?

https://github.com/omichel/webots/blob/0aee13196d44be0aea6232113224f4b8255efb9e/src/webots/core/WbPreferences.cpp#L67
https://github.com/omichel/webots/blob/0aee13196d44be0aea6232113224f4b8255efb9e/src/webots/engine/WbSimulationWorld.cpp#L323

omichel, 2019-03-04:
Unless someone raises a good reason for keeping it, I would remove it.

stefaniapedrazzi, 2019-03-04:
I think it can be removed.
I retrieved a log description from an old commit of Mustafa:

> Log:
ODE_MT 2.0 ---> 1. refactored code in util_MT.cpp to the folder ode_MT. definitions in util_MT.h are now in ode_MT.h 2. Made interfaces for threading and clustering. A Cluster Manager now handles the creation, update and deletion of clusters. 3. A ThreadManager now handles the creation, update and deletion of threads. 4. ThreadManager has a version for both Pthread and OpenMP. 5. Switched threading implementation to OpenMP. 6. If you notice your computer in hyperdrive (all cpus loaded 100% all the time when running webots), please add the OMP_WAIT_POLICY=PASSIVE as an environment variable before you run webots.

DavidMansolino, 2019-04-04:
Fixed in #336

PR #336 (2019-04-04): Cleanup threadingPolicy.
Fix #268.
closed 2019-04-04 by DavidMansolino