cyberbotics/webots #276: GlossyCarPaint and MatteCarPaint are too specific

fabienrohrer, 2019-03-12:
GlossyCarPaint and MatteCarPaint appearances are slightly used in other contexts.
They are basically a flat surface painted with a glossy or matte paint. Or metallized paint?

I propose to simply remove the "Car" keyword from their names.
@all Is it ok like this, or do you have better propositions?

omichel, 2019-03-12:
It seems to refer to paint on a metallic or bodywork surface, which may be different from a wooden or plastic surface... Why not replacing "Car" with "Bodywork"?

DavidMansolino, 2019-03-12:
From the definitions I found, `Bodywork` seems to be related to vehicles only, but these appearances should be much more generic.

fabienrohrer, 2019-03-12:
I think that the matter (if it's flat and without defaults) behind the paint does not affect much the final rendering if the paint cover it perfectly (which is the case here).

omichel, 2019-03-12:
Ok, then, I am fine to remove the "Car" word as initially proposed by @fabienrohrer.

fabienrohrer, 2019-03-12:
👍 
Ok, I will go forward like this.

fabienrohrer, 2019-03-13:
CarLeather suffers from the same issue.

omichel, 2019-03-13:
Same problem, same solution 😄

fabienrohrer, 2019-03-13:
Fixed in #277

PR #277 (2019-03-13): Improve appearance names
Fix #276

- [x] Rename `MatteCarPaint` -> `MattePaint`
- [x] Rename `GlossyCarPaint` -> `GlossyPaint`
- [x] Rename `CarLeather` -> `Leather`
- [x] Update doc accordingly
- [x] Update icons accordingly
- [x] Add missing `Leather` to the icon studio
- [x] Log
closed 2019-03-13 by fabienrohrer