cyberbotics/webots #281: Wrong sphere texture mapping in WREN

stefaniapedrazzi, 2019-03-14:
**Describe the Bug**
The texture mapping for a sphere displayed in Webots doesn't match the X3D texture mapping.

**Steps to Reproduce**
1. Go to 'textured_shapes.wbt'
2. Export HTML5 model
3. Open the exported model in a browser
4. The texture mapping in the browser and in Webots for the sphere are different in the top and bottom area

**Expected behavior**
The sphere texture mapping in WREN should match the X3D one.

**Screenshots**
Webots:
[image]
x3dom:
[image]
view3dscene:
[image]

fabienrohrer, 2019-03-14:
The fix should probably target R2019b instead, because it may have consequences for the users to fix the Webots mapping. Is this true?

stefaniapedrazzi, 2019-03-14:
Yes, i will target 2019b.
I don't think that users will have to change anything, but sensors can behave slightly differently.

stefaniapedrazzi, 2019-03-14:
The mismatch comes from the different triangulation of the sphere:

[image]

The WREN implementation (icosphere) is probably better as all the triangles are about the same size. But it doesn't match the sphere (uv sphere shape used by X3D (x3dom) and THREE.js.

I checked in Blender and both primitives `UV Sphere` and `Icosphere` are available.
If we want to keep the icosphere, then we have to export it as IndexedFaceSet and not as a Sphere geometry.

omichel, 2019-03-14:
Yes, let's do this: export our Sphere primitive as mesh for THREE.js.

DavidMansolino, 2019-04-04:
Linked to #254

stefaniapedrazzi, 2019-04-29:
During the meeting we discussed to add a flag to the Sphere node specifying if the mapping/triangulation type: icosphere or uv-sphere.

* Webots: by default creates an icosphere
* VRML import: it imports a uv-sphere
* VRML/X3D export:
  * uv spheres are exported as Sphere
  * icospheres are exported as IndexedFaceSet

The flag could be called "uv" or "ico"

omichel, 2019-04-29:
I have a small preference for "ico".

stefaniapedrazzi, 2019-04-29:
THREEjs has the `IcosahedronGeometry` that corresponds to the icosphere.
So we could maybe use the Sphere node when exporting to x3d (not VRML) instead of the IndexedFaceSet.

stefaniapedrazzi, 2019-05-14:
Fixed in #407

closed 2019-05-14 by stefaniapedrazzi