cyberbotics/webots #287: libnss3 required on ubuntu 18.04 minimal

ibjeong, 2019-03-16:
**Describe the Bug**
libnss3 is required to run webots on ubuntu 18.04 minimal

**Steps to Reproduce**
1. Install webots with `sudo apt install webots` on ubuntu 18.04 minimal
2. see the message `/usr/local/webots/bin/webots-bin: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory`
3. Install libnss3 `sudo apt install libnss3`
4. Webots runs.

**Expected behavior**
Webots runs without manually installing libnss3

**Screenshots**

    $ webots
    /usr/local/webots/bin/webots-bin: error while loading shared libraries: libnss3.so: cannot open shared object file: No such file or directory
    $ sudo apt install libnss3
    ...
    $ webots # now webots runs

**System**
 - Operating System: Linux Ubuntu 18.04

**Additional context**
I'm trying to run Webots on docker. I don't usually use ubuntu, so honestly I'm not sure if there's Ubuntu 18.04 minimal or not, but I guess you will know what I meant.

DavidMansolino, 2019-03-19:
This is probably because you did install a minimal version of Ubuntu 18.04, I checked on a normal install of Ubuntu 18.04 and libnss3 is present by default.

Did you install Webots from the tarball or debian package?

DavidMansolino, 2019-03-19:
We fixed this in #293 libnss3 will be a dependency automatically installed when installing Webots in the next version of Webots.

PR #293 (2019-03-19): Add libnss3 to Debian Package
Fix #287
closed 2019-03-19 by DavidMansolino