cyberbotics/webots #288: Flashback occurs when using a physical plugin

beatther, 2019-03-18:
**Describe the Bug**
I want to realize that when two robots move forward in my project, they do not collide, but pass through their bodies. The engineer told me that using physical plug-ins may achieve this effect, and let me first familiar with samples / howto / Worlds / physics.wbt, but I just commented on one of the lines of code: 

//dWebotsConsolePrintf("Initialization: ROBOT = %p %p\n", robot_geom, robot_body);

The webots quit directly, and when they re-entered, they executed the project as soon as they entered, and immediately quit.

**System**
 - Windows 10

omichel, 2019-03-18:
I can indeed reproduce this problem. I am investigating and will post back on this issue whenever I better understand what's going on.

omichel, 2019-03-18:
OK, I found the cause of this problem.
Webots is compiled on Windows with gcc-8, but the provided compiler for compiling controllers and physics plugins is gcc-7. Although gcc-7 works fine compiling controllers, for some reason, gcc-7 creates physics plugins (DLL) that cannot be linked dynamically with Webots: Webots crashes when loading these DLLs.

A quick work around for this problem is to install the development environment as explained [here](https://github.com/omichel/webots/wiki/Windows-installation) and to compile the physics plugins from the MSYS2 terminal. It will use gcc-8 by default and hence will work.

We will of course consider providing a better solution to this problem, that will be included to the next revision of Webots.

beatther, 2019-03-18:
After I installed the MSYS2 terminal, when downloading the core components, the download was unsuccessful, which caused my current work to be impossible. So, is there any other way to compile the physical plugin?

omichel, 2019-03-18:
Can't you simply install MSYS2 from [here](https://www.msys2.org/)? If you can do that, simply installing gcc with `pacman -Syuu mingw-w64-x86_64-toolchain` in the MSYS2 console should do the job. Then, you can simply `cd` to the directory where your physics plugin lies and recompile it with `make clean`, `make release`. Unfortunately, there is no other way.

beatther, 2019-03-18:
[image]
Should I install all of them?

omichel, 2019-03-18:
Yes (it will take some time to download and install).

beatther, 2019-03-19:
Thank you very much, according to the method you taught, my problem has been solved temporarily, thank you again!

beatther, 2019-03-19:
um, the unfortunate thing happened again. After I used the function of the physical plug-in several times, there was a flashback situation, and the use of msys compile also flashed back.

omichel, 2019-03-19:
If you modified the physics plugin, it may be possible that you introduced a crash in the source code, causing the crash of Webots. To ensure that you are compiling with gcc-8, you can type `gcc --version` in the MSYS2 console: it should display version `8.3.0`.

beatther, 2019-03-19:
My gcc version is 7.4.0. I am directly importing pacman -S gcc to update the gcc version, but it shows that it is already the latest version. Where should I go to download the 8.3.0 version of gcc?

omichel, 2019-03-19:
gcc-8 comes from `pacman -S mingw-w64-x86_64-gcc`.

beatther, 2019-03-19:
I have solved it.Thank you !!

DavidMansolino, 2019-05-27:
@omichel do you really want to fix this for R2019b ?

omichel, 2019-05-27:
Not sure. If I can't find a simple solution, I will postpone it to R2019b-rev1.

omichel, 2019-06-03:
Fixed in #506.

PR #506 (2019-06-03): Windows: distribute patched gcc-9 instead of gcc-7
Fixes #288 and #175.

I finally found the cause of the bug preventing gcc-8 to link when it was installed in a folder with a space. The problem comes from `windres` which fails at executing in a folder containing a space. `windres` is called at link time by `ld` to link the resulting executable with a file called `default-manifest.o`. If this file is not present, then `ld` doesn't attempt to use it and doesn't fail. It seems this `default-manifest.o` file is not very important and according to my tests, everything works fine without it. See ht [... cut]
closed 2019-06-03 by omichel