cyberbotics/webots #303: VRML Import not Working

DavidMansolino, 2019-03-25:
**Describe the Bug**
VRML exported from CREO are not imported in Webots (whereas they are imported without any error in Blender).

**Steps to Reproduce**
1. Try importing this file:
[x-430_new_asm_prt.wrl.zip](https://github.com/omichel/webots/files/3002054/x-430_new_asm_prt.wrl.zip)
2. Nothing is imported and no error message is printed.

toni-ght, 2020-03-18:
I have the same problem... my .wrl file has exactly the same structure. Do you know a workaround?

DavidMansolino, 2020-03-18:
One workaround is to import it in blender and the use the blender-webots exporter plugin of Blender to export the mesh:
https://github.com/cyberbotics/blender-webots-exporter

ipa-rwu, 2020-04-21:
I also have same issue. 
I import a VRML file(.wrl) to webots (FILE>Import VRML97) 
First I try the VRML file from ```webots/resources/nodes/```, e.g. Sphere.wrl. Then I got error: ``error: Expected field name or '}', found 'vrmlField'``.
Then I try the module my colleague created. Then I got ``error: Skipped unknown 'Separator' node or PROTO``
Does anyone have a solution?

DavidMansolino, 2020-04-21:
Our development version of Webots supports now import of other formats than VRML, I would suggest to try our nightly build beta version of Webots R2020b, you can download it here: 
https://github.com/cyberbotics/webots/releases
Here is the documentation of the new importer:
https://www.cyberbotics.com/doc/guide/the-user-interface?version=develop#file-menu
(see ``Import 3D Model`` item)

ShuffleWire, 2021-06-13:
After so research, I've found out some stuff, using the x-430_new_asm_prt.wrl.zip file as example
Open Webots, Load a empty world, and "File / Import 3D model..."

1) When loading the Node "Header_Info"
WbProtoModel *WbProtoList::findModel(const QString &modelName, const QString &worldPath, QStringList baseTypeList) WbProtoList.cpp:167 nicely find a node (which is correct, you define the corresponding PROTO node above)
So, WbConcreteNodeFactory:284:"WbProtoModel *const model =" give model != NULL, which is fine.
However, line 295:WbNode::createProtoInstance(model, tokenizer, WbWorld::instance() ? WbWorld::instance()->fileName() : "");
yield NULL.
Walking throught that give me : 
WbNode.cpp:1578:WbNode *instance = createProtoInstanceFromParameters(proto, parameters, protoLevel < 1, worldPath); 
which yield instance == NULL
Going into that : 
WbNode.cpp:1618:WbNode *newNode = proto->generateRoot(parameters, worldPath, uniqueId);
which yield newNode == NULL
Going into that : 
WbProtoModel.cpp:325: if (mContent.isEmpty()) 
The condition is actually true, which trigger "return NULL" -> this NULL is then propagated, and cause the node to not be loaded.
I do not know if this is the expected behaviour.

2) When loading the Node "DEF V_Model Anchor"
WbProtoModel *WbProtoList::findModel(const QString &modelName, const QString &worldPath, QStringList baseTypeList) WbProtoList.cpp:167 yield NULL, which cause the whole node to be skipped. As your entire model is contained in this node, nothing is loaded. The fact that no warning are issued is another thing that should be disccused.
Regarding your file, you don't define the PROTO node "Anchor", which seems to cause the issue, as far as I understand. (as this is why findModel() return NULL, afaik)

This is my take on this issue, hope it might help to fix it :) 

Tested on Debian 11, on master at c3c2f47