cyberbotics/webots #317: SUMO not working out of the box on Ubuntu 18.04

stefaniapedrazzi, 2019-03-28:
**Describe the Bug**
SUMO interface doesn't work on Ubuntu 18.04 if Webots is installed from the git repository.
This is because the `netconvert` binary requires some old versions of libraries that are not available on Ubuntu 18.04.

It would be good to detect the OS during the compilation and automatically download and install the required libraries:
* libwebp.so.5
* libjasper.so.1
* libpng12.so.0

**Steps to Reproduce**
1. Start `sumo_interdace_example.wbt`

**System**
 - Operating System: Linux Ubuntu 18.04

DavidMansolino, 2019-04-04:
I can indeed reproduce this, even from my development environment on Ubuntu 18.04.

DavidMansolino, 2019-04-04:
I did fix the sumo package on the ftp.
@stefaniapedrazzi can you confirm that it works fine after a `make clean -C $WEBOTSHOME/projects/default/resources; make -C $WEBOTSHOME/projects/default/resources`?

stefaniapedrazzi, 2019-04-04:
Yes, it works now. Thank you.

closed 2019-04-04 by stefaniapedrazzi