cyberbotics/webots #321: Support in ROS2

talregev, 2019-03-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

The 4th ROS2 distribution is go out in May.   
Can you also support ROS2?

DavidMansolino, 2019-03-29:
This is indeed on our roadmap, we are currently investigating the possibility to support ROS2 in Webots.
Of course, any help is welcome.

DavidMansolino, 2019-06-20:
Just for the note, there are progresses with the support of ROS2 on universal robot side: https://github.com/ros-industrial/ur_modern_driver/issues/244#issuecomment-503068616
And people complaining that meshes are looking bad in gazebo: https://github.com/ros-industrial/universal_robot/issues/425
:wink:

DavidMansolino, 2019-08-22:
Just for the note, we are progressing on this front on this repo: https://github.com/cyberbotics/webots_ros2

DavidMansolino, 2019-09-03:
I believe we can close this issue as it is currently in active development in https://github.com/cyberbotics/webots_ros2

closed 2019-09-03 by DavidMansolino