cyberbotics/webots #323: Hash Sum mismatch on apt install webots

talregev, 2019-03-30:
**Describe the Bug**
A clear and concise description of what the bug is.
When installing webots throw apt install webots I got an error msg:
```bash
Err:2 https://www.cyberbotics.com/debian binary-amd64/ webots 2019a-rev1
  Hash Sum mismatch
  Hashes of expected file:
   - SHA512:b5d5aafb9c2b31ad5622df2417bcf0e4d82e9da0284f0ebf5bf9e6bb4968b3f7ca4b313de259e52e7955ab99ba6ed2044c46ba013b595770c8abc589a9a62864
   - SHA256:e6a9e0c3735de6d240210ed326619d65be744b37d21300625473c3524d531a6d
   - SHA1:3fddc79b1602d7ae2b0692ea4ee640f6eec32477 [weak]
   - MD5Sum:88aad434e888753cc7368d0d3392f5a2 [weak]
   - Filesize:1274420808 [weak]
  Hashes of received file:
   - SHA512:92478212c2a13b017bcddf6e2c63a864de1498643346fdb4073c74ccc3a274c5f8cc48f3addbc3b3b1e97063f13696aa0820d389f7fcd4c60b2cf2e516d69041
   - SHA256:8cd483a21e3667de95caed1a5b5e0aee35accbc50d5b68b611262a02a2e08a1c
   - SHA1:dc109fd3b90244cce3875db3ca0a9598f9addf8a [weak]
   - MD5Sum:957f7c711177563c6eaf8a2df7fd040c [weak]
   - Filesize:1274420808 [weak]
  Last modification reported: Wed, 13 Feb 2019 16:08:52 +0000

E: Failed to fetch https://www.cyberbotics.com/debian/binary-amd64/webots_2019a-rev1_amd64.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA512:b5d5aafb9c2b31ad5622df2417bcf0e4d82e9da0284f0ebf5bf9e6bb4968b3f7ca4b313de259e52e7955ab99ba6ed2044c46ba013b595770c8abc589a9a62864
    - SHA256:e6a9e0c3735de6d240210ed326619d65be744b37d21300625473c3524d531a6d
    - SHA1:3fddc79b1602d7ae2b0692ea4ee640f6eec32477 [weak]
    - MD5Sum:88aad434e888753cc7368d0d3392f5a2 [weak]
    - Filesize:1274420808 [weak]
   Hashes of received file:
    - SHA512:92478212c2a13b017bcddf6e2c63a864de1498643346fdb4073c74ccc3a274c5f8cc48f3addbc3b3b1e97063f13696aa0820d389f7fcd4c60b2cf2e516d69041
    - SHA256:8cd483a21e3667de95caed1a5b5e0aee35accbc50d5b68b611262a02a2e08a1c
    - SHA1:dc109fd3b90244cce3875db3ca0a9598f9addf8a [weak]
    - MD5Sum:957f7c711177563c6eaf8a2df7fd040c [weak]
    - Filesize:1274420808 [weak]
```
**Steps to Reproduce**
curl https://www.cyberbotics.com/Cyberbotics.asc | sudo apt-key add -
sudo apt-add-repository 'deb http://www.cyberbotics.com/debian/ binary-amd64/'
sudo apt update
sudo apt install webots

**Expected behavior**
A clear and concise description of what you expected to happen.
webots should install

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
ununtu 18.04

**Additional context**
Add any other context about the problem here.

omichel, 2019-04-01:
I just tried this and it worked out without any error for me. Can anyone else double-check this?

stefaniapedrazzi, 2019-04-04:
I also checked the installation on Ubuntu 18.04 from the debian repository as described in the user guide and it is working correctly for me as well.

omichel, 2019-04-04:
I am closing this issue as it is not reproducible.
I am assuming it may be caused by a wrong apt configuration or a local network problem.

closed 2019-04-04 by omichel

talregev, 2019-04-04:
I check again and it working. My guess it was something temporary related to hash of the file.

omichel, 2019-04-04:
Thank you for the feedback.