cyberbotics/webots #324: Buggy texture appearance on Linux

stefaniapedrazzi, 2019-04-01:
**Describe the Bug**
Very often when opening a world the textures rendering is buggy or black.
Changing the image quality to medium solves the issue.

**Steps to Reproduce**
Open any world with Image quality High.

**Screenshots**

This is an example, but sometimes the result is much worst and the scene is unrecognisable.
[image]

**System**
 - Operating System: Linux Ubuntu 18.04 (Linux 4.15.0-46-generic x86_64)
 - Graphics Card:
    - OpenGL vendor: NVIDIA Corporation
    - OpenGL renderer: GeForce GT 740/PCIe/SSE2
    - OpenGL version: 3.3.0 NVIDIA 390.116
    - Available GPU memory: 2097152 bytes

DavidMansolino, 2019-11-28:
It would be very interesting to know if the bug is still visible on develop.
I am suspecting that the previous background implementation was causing many side effect bug.

stefaniapedrazzi, 2019-11-28:
Unfortunately I had this bug with the old PC.
At the moment, my old graphic card is not installed anywhere, maybe I can build a new PC using some old components but I have to check if I have all what is needed.

fabienrohrer, 2019-11-28:
I think this kind of bug should be fixed when we have them under the eyes, the cost to fix it now is too big in my point of view. I would close it. What do you think?

stefaniapedrazzi, 2019-11-28:
These are bugs that are not reproducible every time.
Investigating them takes a lot of time just to be able to reproduce them and most of the time doesn't have a high fix priority
For me the point of these issue report is to log somewhere that this happens and inspect if happens just on one setup or more.
Even for the future it is good to know that there was this issue.

Anyway i agree that at this point this issue can be closed.

closed 2019-11-28 by stefaniapedrazzi