cyberbotics/webots #329: Mismatch between PROTO list and Addnode Dialog

DavidMansolino, 2019-04-02:
**Describe the Bug**
A new PROTO can be added using the add node dialog but then it is not found at revert.

**Steps to Reproduce**
1. Start Webots.
2. Create a new PROTO file.
3. Add this new PROTO node to the simulation => PROTO found.
4. Save and revert the simulation => PROTO not found.

**Expected behavior**
It should find the PROTO even after revert.

DavidMansolino, 2019-04-02:
It is actually only if the new PROTO is located in the installation directory of Webots, because the WbProtoList assumes that the files in the installation directory won't change during runtime.

DavidMansolino, 2019-04-02:
Not sure what is the best solution, to re-generate completely the list at revert or to deny adding them from the add node dialog.

omichel, 2019-04-02:
I would avoid re-generating the complete list for just this very special case.

DavidMansolino, 2019-04-02:
I checked and both would complexify the code and increase either the world loading time or add node creation. I propose to keep it as is as it's a shame to slow down one or the other for this very specific edge case.

closed 2019-04-02 by DavidMansolino