cyberbotics/webots #330: Buggy rendering order with ElevationGrid

stefaniapedrazzi, 2019-04-02:
**Describe the Bug**
Some parts of the ElevationGrid are wrongly rendered above others.

**Steps to Reproduce**
1. Go to "tests/api/worlds/pen_elevation_grid.wbt"
2. Change the BOARD shape texture to make it more visible.

**Screenshots**

[image]

**System**
 - Operating System: Linux Ubuntu 18.04
 - Graphics Card: NVIDIA  GeForce GT 740/PCIe/SSE2

fabienrohrer, 2019-04-08:
This is because `color_checker_chart.png` has an alpha channel. The shape is rendered without depth check, so the polygon order maters.

@stefaniapedrazzi Is it a sufficient answer? Could we close this issue?

stefaniapedrazzi, 2019-04-08:
Thank you for the explanation but this is in any case a bug.
In THREEjs this object is rendered correctly so this is a specific issue of WREN.

omichel, 2019-04-08:
If that's a bug, we shouldn't close the issue. Can't we simply simply remove the alpha channel from this texture image which doesn't seem to need it?

omichel, 2019-04-08:
PNG images should be used only when we need an alpha channel, as they are usually bigger than their equivalent JPEG images. So, any PNG texture that doesn't need transparency or alpha blending should be converted to a JPEG texture.

stefaniapedrazzi, 2019-04-08:
> If that's a bug, we shouldn't close the issue. Can't we simply simply remove the alpha channel from this texture image which doesn't seem to need it?

This was just a test world created to check the ElevationGrid texture mapping and I picked the first image I found without checking if it had alpha channel or not.

fabienrohrer, 2019-04-08:
To generalize the discussion, I think this is not a bug but an implementation choice. It's a matter of how we sort the polygons in case of a transparent material. Currently we do nothing to save resources. It's not a bad choice.

About JPEG vs PNG, I completely agree to remove alpha when unneeded, but the present image is a typical example where the PNG format (3Ko) has a smaller size than the JPEG format (30Ko) because of its big homogenous areas. JPEG should be used for real pictures rather.

stefaniapedrazzi, 2019-05-27:
The alpha channel in `color_checker_chart.png` has been removed.

PR #355 (2019-04-15): Removed useless alpha channels, optimized PNG file sizes
Fixes #330.
Several PNG files were containing a useless alpha channel.
closed 2019-05-27 by stefaniapedrazzi