cyberbotics/webots #358: Issues applying a force to the e-puck from the scene tree

stefaniapedrazzi, 2019-04-11:
**Describe the Bug**
Depending on where you click the apply force functionality doesn't work: it seems there are issues when determining if it is a static or dynamic object.

**Steps to Reproduce**
1. Go to e-puck.wbt
2. Start the simulation
3. Apply a force by clicking on the a wheel -> it works correctly
4. Apply a force by clicking on the transparent external case -> it doesn't work

**Expected behavior**
It should be possible to apply a force to a Solid with physics by clicking on any part of it.

ad-daniel, 2020-12-28:
It isn't technically a bug, the e-puck's "body/chassis" is actually a huge LED.

     DEF EPUCK_BODY_LED LED {
      rotation 0 1 0 4.712399693899575
      children [
        Shape {
          appearance PBRAppearance {
            baseColor 0.5 0.5 0.5
            transparency 0.4
            roughness 0.5
            metalness 0
            emissiveIntensity 0.2
          }
          geometry IndexedFaceSet {

            [ ... POINTS & COORDS ...  ]

            creaseAngle 0.5
          }
        }
      ]
      name "led8"
      color [
        0 1 0
      ]
    }

Being a SolidDevice it doesn't have physics enabled by default hence why it can't be pulled from there. The e-puck docs mention that its body can light up so that might explain the choice of modeling it like this.

Enabling the physics in the body is kinda awkward though because the boundingObject and mass/inertia of the entire e-puck is enforced at the [Robot level](https://github.com/cyberbotics/webots/blob/3766a52ca250cc6ba41a777182a4aeb61d2d3642/projects/robots/gctronic/e-puck/protos/E-puck.proto#L1144-L1177). So it requires giving this LED a negligible mass and a dummy boundingObject  like this:

      boundingObject Transform {
        translation 0.0 0.03 0.0
        children [
          Box {
            size 0.001 0.001 0.001
          }
        ]
      }
      %{ if kinematic == false then }%
      physics Physics {
        density -1
        mass 0.0001
      }
      %{ end }%

Whether that's desirable I don't know.

stefaniapedrazzi, 2021-01-04:
We don't want to change the `e-puck` robot model.

Instead we should evaluate if it is worth changing the method to apply force/torque so that it goes up the tree until it finds a Solid with physics. Currently it stops at the first ancestor `Solid` of the picked geometry.

closed 2021-01-07 by ad-daniel