cyberbotics/webots #361: Too Big Images in Docs

DavidMansolino, 2019-04-12:
Some of the images in the doc have way too high resolutions and slow down the page loading with slow/medium connections. This is particularly visible in the samples and robots pages.
For example on this page: https://cyberbotics.com/doc/guide/samples-devices
There are a lot of big images which are actually displayed very small.

We should probably keep this big images for the future when we will support modal images, but for the preview we should use smaller images.

fabienrohrer, 2019-04-15:
My original plan was to create an automatic thumbnail mechanism and the possibility to see the original images in a modal window.

DavidMansolino, 2019-04-15:
That would be indeed a cleaner mechanism, but I think we should do this urgently (as last week when I was showing to the doc to potential clients it was giving a bad image of the doc to have it loading slowly).

closed 2019-05-16 by DavidMansolino