cyberbotics/webots #362: Add the possibility for loading controllers given their path

schra, 2019-04-12:
**Is your feature request related to a problem? Please describe.**

I have the following situation: I'm building my controllers with a custom `CMakeLists.txt`. Thus the executables of the controllers will end up in the `build` directory. I also have a `.wbt` file that associates objects with controllers.

Now to the problem: I want to call `webots myworld.wbt` (which in turn should load my controllers) without being required to have the folder structure as described [here](https://cyberbotics.com/doc/guide/controller-start-up) with the `controllers/` and `worlds/` folder. Creating such a folder structure within my  `CMakeLists.txt` file would either require me to `install()` the files in the correct folder structure (and therefore forcing me to always call `make install`) or to resort to some nasty CMake or Bash hacks.

**Describe the solution you'd like**

Therefore I propose to add the possibility for loading controllers given their path. I came up with the following ideas that would implement this proposal:

* Introducing the command line parameter `--controller-folder` that sets the folder where the executable is searched in.

    For example, if I specify `controller "mycontrollername"` in my `.wbt` file and pass `--controller-folder /path/to/build/` then Webots should try to load the executable `/path/to/build/mycontrollername`. Note that the path is *not* `/path/to/build/mycontrollername/mycontrollername`.
* Controllers could additionally be searched in the current working directory (`$PWD`), or within your path (`$PATH`).

    For example, if I specify `controller "mycontrollername"` in my `.wbt` file and let my current working directory be `/path/to/build` then the controller could be searched in the folders that it's already currently searched in according to the current implementation and after that the controller could be searched in `/path/to/build/mycontrollername`. Also note here that the path is not `/path/to/build/mycontrollername/mycontrollername`.

I'm more than happy to provide a patch for this. With this issue I want to ask if you are open to accept such a patch and which of the solutions you prefer, or if you have any other ideas for fixing this problem?

omichel, 2019-04-15:
Thank you for this proposal which looks very interesting. We would actually prefer the second option with the `PATH` environment variable because it seems simpler and cleaner. But I see two problems with this variable: if ever a controller is named the same as an existing command in the `PATH`, it may fail at starting the correct one (e.g., for example a `cat` controller for a kitty robot may launch the UNIX `cat` command line). Also, I fear "polluting" the `PATH` variable which is used by the system for other purposes (e.g., if your controller wants to launch an external process, or needs a DLL on Windows, it will use the `PATH` environment variable). Therefore I would suggest you to use a different environment variable, e.g., for example `WEBOTS_CONTROLLER_PATH` which may contain a list of folders where to search robot controllers if not found at the usual location.

We would be indeed happy to review and merge such a contribution.

So, you are welcome to create a new pull request about it.

schra, 2019-04-15:
I've looked into the code that loads controllers and I have a question about it. My proposal requires that the folder of the controller doesn't have to be named like the controller (see above examples). Currently this is enforced throughout the code in the sense that there is a one "controller directory" that is used to load controllers from:
https://github.com/omichel/webots/blob/3967c5b035b9e74c56136133d6189c0ed10bc19b/src/webots/nodes/WbRobot.cpp#L460
Do you have a proposal how to adjust the code in order to additionally support loading from folders that not necessarily are "controller directories"? It seems that this would require some refactoring, if I'm not mistaken?

> We would actually prefer the second option with the PATH environment variable because it seems simpler and cleaner.

Would you also accept an implementation via command arguments? They fit a bit better in my tooling.

> Therefore I would suggest you to use a different environment variable

I totally agree that using `WEBOTS_CONTROLLER_PATH` is better than using `PATH` or `PWD`.

omichel, 2019-04-15:
Yes, some refactoring may be needed to implement this feature.
Because it is a rare use case, we would prefer the environment variable solution and avoid the command line argument option for now.

DavidMansolino, 2019-08-07:
This is not on our medium-term roadmap.

closed 2019-08-07 by DavidMansolino