cyberbotics/webots #363: [bug] Simulation and session servers

talregev, 2019-04-14:
**Describe the Bug**
Cannot run webots server

**Steps to Reproduce**
1. `cd /usr/local/webots/resources/web/server`
2. `sudo server.sh start`
server.sh: line 46: ./session_server.py: Permission denied
server.sh: line 51: ./simulation_server.py: Permission denied

The script it also run first the session_server.py and not the simulation_server.py,
as recommended in` /usr/local/webots/resources/web/server/README.md`

1. sudo python simulation_server.py config/session/local.json
```python
Traceback (most recent call last):
  File "simulation_server.py", line 20, in <module>
    from pynvml import nvmlInit, nvmlShutdown, nvmlDeviceGetHandleByIndex, nvmlDeviceGetName, nvmlDeviceGetMemoryInfo, \
ImportError: No module named pynvml
```

Switch to python3:
1. `sudo python3 simulation_server.py config/session/local.json`
```python
  File "simulation_server.py", line 78
    os.chmod(fullname, 0775)
                          ^
SyntaxError: invalid token
```
edit line 78: to ' os.chmod(fullname, 775)'
run again 
1. `sudo python3 simulation_server.py config/session/local.json`
```python
Traceback (most recent call last):
  File "simulation_server.py", line 20, in <module>
    from pynvml import nvmlInit, nvmlShutdown, nvmlDeviceGetHandleByIndex, nvmlDeviceGetName, nvmlDeviceGetMemoryInfo, \
ImportError: No module named pynvml
```

**System**
 - Operating System: ubunu 18.04

stefaniapedrazzi, 2019-04-15:
The simulation server works fine because we are daily using it.
But we miss some documentation for it.

Currently it works only for Python 2.7, but I will fix it so that it works for Python3 too.

About the dependencies errors, here are listed the required dependencies:
https://github.com/omichel/webots/wiki/Linux-Optional-Dependencies#webots-web-service

> server.sh: line 46: ./session_server.py: Permission denied
server.sh: line 51: ./simulation_server.py: Permission denied

We already fixed the execution permissions here: #229.
So they should now work correctly. Is your environment up-to-date?

> The script it also run first the session_server.py and not the simulation_server.py,
as recommended in /usr/local/webots/resources/web/server/README.md

It doesn't matter the starting order, and it works correctly independently if the `session_server.py` is started before the `simulation_server.py`.

> There is no key dir inside server.

Yes, because you have to use your own key specific to your website.

> going on web browser on localhost:2000
404: Not Found
going on web browser on localhost:1999
Can "Upgrade" only to "WebSocket".

The simulation server is just a utility that allows your web page to connect to Webots and stream a simulation but it doesn't work by its own.
This means that you have to start it to prepare the Webots environment to receive the streaming requests, but then you also have to build your own web page from which the user can start the simulation.
We also provide the webots.js  library to handle the simulation streaming:
https://github.com/omichel/webots/blob/master/resources/web/wwi/webots.js

If you simply want to stream the current Webots simulation without opening a new Webots instance for each connected user I would suggest you to look at the `streaming_server` example:
https://github.com/omichel/webots/tree/master/resources/web/streaming_viewer

talregev, 2019-04-15:
I want something more simple then `streaming_server`.
Thanks for your info.

closed 2019-04-15 by talregev