cyberbotics/webots #379: Windows build: Requested Qt version not available

fabienrohrer, 2019-04-23:
**Describe the Bug**

When following Windows instructions, the `msys64_installer.sh` script fails on this issue:

```
Qt5 version requested: 5.12.2
Qt5 version installed: 5.10.1-2
Qt5 version available: 5.12.3-1
Requested version not available
```

**Steps to Reproduce**

From a clean Windows install, follow the build instructions of the wiki until executing `msys64_installer.sh`.

- The script seems to do a lot of things, among other related with Qt.
- `./msys64` is not created.
- A user reports the same on Discord ("el").

**Screenshots**

[image]

omichel, 2019-04-23:
I believe we should fix it on the `master` branch. I can take care of it.

fabienrohrer, 2019-04-23:
Ok thank you. Yes, it's clearly an `hotfix` in my point of view too.

PR #380 (2019-04-23): Updated Qt version on Windows
Fixes #379
closed 2019-04-23 by omichel