cyberbotics/webots #384: Webots cannot start on Windows with current MSYS2

omichel, 2019-04-24:
**Describe the Bug**
Webots cannot start if compiled on Windows with the current MSYS2.

**System**
 - Operating System: Windows 10

**Additional context**
The workaround is simple: copy libicu*62.dll from an installed Webots R2019a revision 1 to the development environment: cp /C/Program\ Files/Webots/msys64/mingw64/bin/libicu*62.dll ~/webots/msys64/mingw64/bin
The problem was reported here: https://github.com/msys2/MINGW-packages/issues/5273 and should be fixed soon hopefully.

fabienrohrer, 2019-04-24:
For conveniency, I created an archive with the files to copy:
[libicu.zip](https://github.com/omichel/webots/files/3112115/libicu.zip)

omichel, 2019-04-26:
Fixed. See https://github.com/msys2/MINGW-packages/issues/5273

closed 2019-04-26 by omichel