cyberbotics/webots #386: Webots crashes

tilly111, 2019-04-24:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: [e.g., Windows 10, Linux Ubuntu 18.04, macOS Mojave]
 - Graphics Card: [e.g., NVIDIA GeForce RTX 2080 11 GB, AMD Radeon RX 580 8GB, etc.]

**Additional context**
Add any other context about the problem here.

DavidMansolino, 2019-05-02:
Thank you for this crash report,
I can indeed reproduce it (on Linux too).
Webots does crash with the following stack:
```
#0  0x00000000009230b1 in WbPaintTexture::pickColor (this=0x9538d90, 
    pickedColor=..., uv=...) at nodes/utils/WbPaintTexture.cpp:275
#1  0x0000000000999fee in WbShape::pickColor (this=0x6f32190, pickedColor=..., 
    ray=...) at nodes/WbShape.cpp:316
#2  0x0000000000834eca in WbDistanceSensor::computeValue (this=0x7f2b000)
    at nodes/WbDistanceSensor.cpp:521
#3  0x0000000000834a69 in WbDistanceSensor::refreshSensorIfNeeded (
    this=0x7f2b000) at nodes/WbDistanceSensor.cpp:473
#4  0x0000000000835651 in WbDistanceSensor::writeAnswer (this=0x7f2b000, 
    stream=...) at nodes/WbDistanceSensor.cpp:601
#5  0x000000000097a982 in WbRobot::dispatchAnswer (this=0x7c305f0, stream=..., 
    includeDevices=true) at nodes/WbRobot.cpp:977
#6  0x000000000080be88 in WbController::writeAnswer (this=0x65ddec0, 
    immediateAnswer=false) at control/WbController.cpp:893
#7  0x0000000000801085 in WbControlledWorld::waitForRobotWindowIfNeededAndCompleteStep (this=0x6e15670) at control/WbControlledWorld.cpp:384
#8  0x0000000000800519 in WbControlledWorld::step (this=0x6e15670)
    at control/WbControlledWorld.cpp:287
#9  0x00000000007ffe83 in WbControlledWorld::processWaitingStep (
    this=0x6e15670) at control/WbControlledWorld.cpp:208
#10 0x00000000007ffe41 in WbControlledWorld::triggerStepFromTimer (
    this=0x6e15670) at control/WbControlledWorld.cpp:202
```

DavidMansolino, 2019-05-02:
As a temportary workaround you can fix it by removing the `Thymio2Pen` nodes in the `bodySlot` fields of the `Thymio2` robot nodes.

DavidMansolino, 2019-05-02:
This bug was fixed in #412, the fix will be included in the next release of Webots.

PR #412 (2019-05-02): Fix DistanceSensors with Pen
Fix  #386
closed 2019-05-02 by DavidMansolino