cyberbotics/webots #387: Cubemap orientation

fabienrohrer, 2019-04-25:
Following #375, the background orientation (and so environment map reflections) in THREEjs and Webots are rotated by 180° vertically. This is currently solved in JS by flipping the top/bottom images and inverting the right/left and front/back textures.

We could explore now a better solution: to break the Webots background orientation.
First step is to determine our compliance with X3D to determine if this is a good idea or not.

fabienrohrer, 2019-04-25:
The Webots orientation does respect the VRML and X3D Background definition, which is good:

As a proof ("right" == `+x`, "back" == `+z`):

[image]
[image]

Therefore, it's a good idea to keep the Webots background orientation as-is, otherwise we will have future issues.

So to improve this topic, it remains only the possibility to modify THREEjs.
I will start to ask a question on to the THREEjs Github issues.

fabienrohrer, 2019-04-25:
Follow up: https://github.com/mrdoob/three.js/issues/16328

fabienrohrer, 2019-05-21:
I tried to put hands on the `three.js` sources.
I can change the background and reflection cubemap orientation with this patch:

```diff
diff --git a/src/renderers/shaders/ShaderChunk/envmap_physical_pars_fragment.glsl.js b/src/renderers/shaders/ShaderChunk/envmap_physical_pars_fragment.glsl.js
index d319e9b88..27dd325a1 100644
--- a/src/renderers/shaders/ShaderChunk/envmap_physical_pars_fragment.glsl.js
+++ b/src/renderers/shaders/ShaderChunk/envmap_physical_pars_fragment.glsl.js
@@ -72,7 +72,8 @@ export default /* glsl */`
 
                #ifdef ENVMAP_TYPE_CUBE
 
-                       vec3 queryReflectVec = vec3( flipEnvMap * reflectVec.x, reflectVec.yz );
+                       mat3 m = mat3(1.0, 0.0, 0.0, 0.0, 0.0, -1.0, 0.0, 1.0, 0.0);
+                       vec3 queryReflectVec = m * vec3( flipEnvMap * reflectVec.x, reflectVec.yz );

diff --git a/src/renderers/shaders/ShaderLib/cube_frag.glsl.js b/src/renderers/shaders/ShaderLib/cube_frag.glsl.js
index 0a53ba08e..dc7ba4699 100644
--- a/src/renderers/shaders/ShaderLib/cube_frag.glsl.js
+++ b/src/renderers/shaders/ShaderLib/cube_frag.glsl.js
@@ -6,8 +6,8 @@ uniform float opacity;
 varying vec3 vWorldDirection;
 
 void main() {
-
-       vec4 texColor = textureCube( tCube, vec3( tFlip * vWorldDirection.x, vWorldDirection.yz ) );
+       mat3 m = mat3(1.0, 0.0, 0.0, 0.0, 0.0, -1.0, 0.0, 1.0, 0.0);
+       vec4 texColor = textureCube( tCube, m * vec3( tFlip * vWorldDirection.x, vWorldDirection.yz ) );
```

[image]

stefaniapedrazzi, 2019-05-21:
I just checked and there is no Cubemap orientation issues when using an equirectangular image (default backgrounds in Webots).

fabienrohrer, 2019-05-21:
Great.

stefaniapedrazzi, 2019-05-21:
Or maybe not, I'm still checking..

fabienrohrer, 2019-05-21:
This information is important. I could also fix it if I do a threejs patch.

stefaniapedrazzi, 2019-05-21:
I'm still having issues applying the background so it doesn't work in webots.min.js, but only in a separate three.js example that doesn't work the same for the moment.
I will confirm if it is correct when it will fully work.

stefaniapedrazzi, 2019-05-22:
Also for HDR equirectangular background, the textures we distribute are mirrored horizontally when applied to the three.js scene.
HDR equirectangular textures from the three.js repository are rendered correctly.
In this case we have to change the shaders defined in the `EquirectangularToCubeGenerator.js` example source file and it would also be easier to patch it locally given that it has to be minified and bundled within webots.min.js.

Note that HDR images always need to be flipped vertically to be displayed correctly.

BenjaminDeleze, 2021-11-15:
Fixed by using wrenjs instead of threejs #2769

closed 2021-11-15 by BenjaminDeleze