cyberbotics/webots #388: Adding e-puck's missing sensors full-support

Thelm76, 2019-04-25:
By testing the e-puck sensors in the webots' simulation, I discovered that there are missing ones in the simulation, like the gyro, the ToF sensor or the magnetometer. 

As I wouldlike to use them with e-pucks, I was wondering wether I can add them in a "simple" way, and made them to work in the simulation and in the real world.

So I started by editing the proto file in order to add these sensors in the simulation. Now, I'm trying to include the values in the communication by editing concerned files in "e-puck\plugin\e-puck_wifi" 

Next, I will look at the files to make programs working directly on the robot (without wifi/bluetooth communication)

As I'm quite a begginer in C/C++ and I started webots like three weeks ago, I think will need some help in order to achieve the implementation of the sensors.

fabienrohrer, 2019-04-25:
Don't hesitate to ask specific questions here, we will help you :-)

Thelm76, 2019-04-25:
I'm facing two problems for now : in this piece of code (e-puck_wifi/Wrapper.cpp) I don't understand the calibration values for the accelerometer so I can't transpose them to the gyro and the magnetometer

```
TripleValuesSensor *accelerometer = DeviceManager::instance()->accelerometer();
  if (accelerometer->isEnabled()) {
    const double calibration_k[3] = {-9.81 / 800.0, 9.81 / 800.0, 9.81 / 800.0};
    const double calibration_offset = -2000.0;
    double values[3];
    for (int i = 0; i < 3; i++) {
      short int acc = sensor_data[2 * i] + 256 * sensor_data[2 * i + 1];
      values[i] = calibration_k[i] * (acc + calibration_offset);
    }
    wbr_accelerometer_set_values(accelerometer->tag(), values);
  }
```

also, on the wiki, I've seen that the accelerometer and the gyro are encoded on 6 bytes so 2 bytes for X, Y and Z. But the magnetometer is encoded on 12 bytes, which doesn't make sense to me as the raw values are between -32460 and 32760... So are there 3 axis encoded on 4 bytes, or am I missing something?

fabienrohrer, 2019-04-25:
About the calibration, it's a matter to convert the e-puck accelerometer hardware value to the Webots metric acceleration force.
A linear function is applied to scaled and shit each component.
The term "calibration" is not very good here, it's more a matter to shift and scale the values by this linear function.

The code is indeed a bit cryptic.
The e-puck hardware returns 3 set of 2 bytes to determine the accelerometer value.
These 2 bytes defines a value between 0 and 4096, where 2000 is the 0 point, and 2800 equals to a force of 9.81.

So this piece of code take these 2 bytes (as a short int), subtract 2000, divide it by 800, and multiply it by 9.81. That's it we have the metric value.

fabienrohrer, 2019-04-25:
To know how each device is encoded, you should read the specifications of these devices! => Yes, it's possible that the magnetometer has a better resolution.

Thelm76, 2019-04-25:
Oh that's why I didn't understand the values... on the wiki they said that the accelerometer raw values are between -1500 and 1500 with a resolution of +-2g... I'll look further into the sensor documentation to figure out which values to put in the calibration

fabienrohrer, 2019-04-25:
If I remember well, I have done this calibration by trial and errors based on experiments :-D

Thelm76, 2019-04-25:
So if I stick with the documentation values I'm not sure it will work well in webots? And as I can't test it on a real e-puck for now, I can't really determine these values?

fabienrohrer, 2019-04-25:
Therefore, reading the real hardware specifications of the devices is the best approach. And reading some examples about the e-puck 2 devices, like is also a good idea. Did you found some?
To be honest, it is difficult to achieve such goal without having access to a real robot :-/

Thelm76, 2019-04-25:
So if I understood the way the file works, I have for example this code for the magnetometer in the wrapper :

```
  TripleValuesSensor *magnetometer = DeviceManager::instance()->magnetometer();
  }
  if (magnetometer->isEnabled()) {
    const double calibration_k[3] = {4912 / 32760, 4912 / 32760, 4912 / 32760}; //TODO determine values
    const double calibration_offset = 0; //TODO determine value
    double values[3];
    for (int i = 0; i < 3; i++) {
      short int magnetometer = sensor_data[24 + 2 * i] + 256 * sensor_data[25 + 2 * i]; //TODO confirm values
      values[i] = calibration_k[i] * (magnetometer + calibration_offset);
    }
    wbr_gyro_set_values(magnetometer->tag(), values);
  }
```

with the 4912 / 32760 corresponding with the value of 32760 when there is a magnetic field of 4912µT (the sign is to be determined with tests) ? and after I think I have to modify the lookup table of the sensor in the proto file, is that right?

Also, I will have e-pucks on hand in a few days I'm just preparing webots in case I need one of the missing sensors. and I can take a look into the gctronic's monitoring program for e-puck2 but it uses an USB protocol.

fabienrohrer, 2019-04-25:
That looks quite good.

I'm wondering if you should not take 4 bytes instead of 2 here!?

```
unsigned int magnetometer = sensor_data[4 * i] + (sensor_data[4 * i + 1] << 8) + (sensor_data[4 * i + 2] << 16) + (sensor_data[4 * i + 3] << 24);
```

fabienrohrer, 2019-04-25:
750 is certainly a better value than 800 because scientifically found.

Thelm76, 2019-04-25:
for the 4 bytes I don't really know as the IMU documentations always says 16 bits and never 32 bits for the magnetometer... and the -32760 - 32760 range can be encoded on 16 bits so 2 bytes seems to be the good value
the e-puck2 TCP packet is described here :
http://www.gctronic.com/doc/index.php?title=e-puck2_PC_side_development#WiFi_2

and the output of the sensor is here in page 50/55 : 
http://projects.gctronic.com/epuck2/doc/MPU-9250-Register-Map.pdf

So I really don't understand why gctronics allocate 12 bytes for the values :/

fabienrohrer, 2019-04-25:
For sure gctronic knows much about the e-puck than us :-/

Thelm76, 2019-04-25:
How! I think I now understand the 2000 in calibration, in fact it would be more logical for it to be 2048 as it is the half of 4096, and as it, 9.81 should corespond to exactly 2798 (2048+750)! but the offset should be 32768 for the accelerometer and the gyro sensors (2^16/2?). I hope I will have an e-puck soon to test this

Thelm76, 2019-04-26:
I think I've found the answer in another piece of code by @omichel there are these few lines in a comment

```
          printf("Acc: %d %d %d\n", *(short int *)(&sensor[0]), *(short int *)(&sensor[2]), *(short int *)(&sensor[4]));
          printf("Acceleration: %g\n", *(float *)(&sensor[6]));
          printf("Orientation: %g\n", *(float *)(&sensor[10]));
          printf("Inclination: %g\n", *(float *)(&sensor[14]));
          printf("Gyro: %d %d %d\n", *(short int *)&sensor[18], *(short int *)(&sensor[20]), *(short int *)(&sensor[22]));
          printf("Magnetometer: %g %g %g\n", *(float *)(&sensor[24]), *(float *)(&sensor[28]), *(float *)(&sensor[32]));
 
```

Thelm76, 2019-04-26:
I think I am thinking the wrong way as the other sensors in webots outputs directly the binary values and in the protofile, each value in the simulation world is converted in the binary equivalent of the sensor's output no? So any "calibration" of sensors is to be done in the controller and not in the wrapper... I think? With this, I just have to make the lookup tables outputs the values of the sensors. e.g for the accelerometer the lookup table should be [ -19.62 -1500 0.0039 16.62 1500 0.0039 ]

Thelm76, 2019-05-02:
Would the added sensors break the compatibility with e-puck v1/1.3 (I mean in the e-puck library for cross-compilation) ?
Cause I am now thinking about living the original e-puck files unmodified, and adding a new e-puck2 robot folder.
I think it would be great to stay with one robot for the two version but I don't know if this is feasible...

fabienrohrer, 2019-05-02:
It depends where and how you add your modifications :-)
In the PROTO, you can encapsulate this thanks to a Lua test (please refer to the EPuck.version field). Wifi library is only used by the e-puck 2. I don't think you changed the firmware, right?

=> If you open a Pull Request if your current code, it would be easier to answer you case by case.

Thelm76, 2019-05-02:
For now, the lua tests are working fine in the proto file, and I made a v2 version of the Bluetooth with a field in the PROTO to switch between WiFi and Bluetooth.
As I am waiting for an e-puck in order to do tests, I wanted to edit the required files to make the programs working on real e-puck without connection to webots. (so I think the firmware and libepuck?)

For the pull request I already made a branch but it's quite messy... And I though that the pull request has to be done when the build is working.

DavidMansolino, 2020-01-20:
As this thread is quite old without news I allow myself to close it.

closed 2020-01-20 by DavidMansolino