cyberbotics/webots #409: ElevationGrid does not update boundingobject

nisuunn0, 2019-05-01:
Bug description
When an elevationGrid is defined, it works fine with respect to collision. However, when a supervisor is used to change the height parameters of the elevationGrid, the visual appearance is updated correctly, but the elevationGrid's collision no longer works properly. The elevationgrid seems still be using its old bounding object, rather than the new one which should follow the updated height values. I also noticed, that before changing height parameters with the supervisor, if you change the position of the elevationgrid and then apply the supervisor's height update, then the problem is no more present and the elevation grid's collision works as expected.

Steps to reproduce
Using a supervisor, change the height field parameters of an already defined elevationgrid and then have something collide with the elevationgrid.

**Expected behavior**
I would expect the collision of the elevationgrid to work with the new height values from the supervisor.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Linux Ubuntu 18.04
 - Graphics Card: 

**Additional context**
Add any other context about the problem here.

DavidMansolino, 2019-05-23:
I can't reproduce this on the revision branch with Ubuntu 16.04, do you have a step by step procedure (or even better a simulation) to reproduce the problem?

@omichel have you been able to reproduce the problem?

DavidMansolino, 2019-05-23:
Ok, I have finally been able to reproduce the issue. The problem is not that the actual physical geometry is not updated but the 'new touching' solids are not enabled.

DavidMansolino, 2019-05-23:
Here is the problematic part of the code:
https://github.com/omichel/webots/blob/4f07a74a0631952a7b098e99f0ab92878051e256/src/webots/engine/WbSimulationCluster.cpp#L703
We don't create contact joint between kinematic and disabled object, but here we are exactly in this case, the eleveationGrid is kinematic and the Solid is disabled (sleeping).

DavidMansolino, 2019-05-23:
That's not the only problem, because removing this condition doesn't resolve the problem (contact points are correctly detected now but solid remains disabled).
And the problem is not strictly related to elevationGrid,  any kinematic solid whose size change can reproduce the problem when colliding with a dynamic solid **with joints**.

DavidMansolino, 2019-05-23:
Calling `wb_supervisor_simulation_reset_physics` in the supervisor just after changing the elevationGrid height is a working temporary workaround as it makes sure that all the solid are enabled.

PR #494 (2019-05-28): Fix Geometry Update not Awakening Solids
Attempt to fix #409
closed 2019-05-28 by DavidMansolino