cyberbotics/webots #423: Conflicting dependencies on Linux

stefaniapedrazzi, 2019-05-06:
On Ubuntu 18.04 when installing the `npm` package required to install babel and create `webots.min.js` the following packages are removed:
* `libssh-dev`
* `libssh-dev`

and these new packages are installed:
* `libssl1.0-dev`
* `node-gyp`
* `nodejs-dev`
* `npm`

as a consequence the there are compilation issues with the RobotisOP2 robot window:

```
In file included from TransferWidget.cpp:16:0:
SSH.hpp:25:10: fatal error: libssh/callbacks.h: No such file or directory
 #include <libssh/callbacks.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/stefi/develop/webots/resources/Makefile.include:679: recipe for target 'build/debug/TransferWidget.o' failed
```

We need to fix the compilation and update the installation instructions.

stefaniapedrazzi, 2019-05-06:
Maybe the compilation issues are fixed by a `make clean`, I'm checking.
But the installation instructions should be improved as it doesn't make sense to install dependencies and remove them later.

stefaniapedrazzi, 2019-05-06:
No, even after `make clean` I get the same compilation error.

stefaniapedrazzi, 2019-05-06:
This is a known issue of nodejs v8 (current default version).
Installing nodejs v10 fixes the issue as the dependencies have been fixed.

fabienrohrer, 2019-05-06:
Well spotted!

stefaniapedrazzi, 2019-05-06:
I just updated the installation instructions:
https://github.com/omichel/webots/wiki/Linux-Optional-Dependencies

closed 2019-05-06 by stefaniapedrazzi