cyberbotics/webots #433: Upgrade the robot components to the new webots.min.js

fabienrohrer, 2019-05-07:
Porting the robot components to the new THREEjs viewer becomes important.

It's currently broken in some branches:
https://github.com/omichel/webots/pull/430

stefaniapedrazzi, 2019-05-08:
On develop I'm also getting this error in the terminal when starting Webots:
```
QString::arg: Argument missing: (%s:%u, %s), file:///home/stefi/webots/docs/dependencies/wwi/R2019b/webots.min.js
QString::arg: Argument missing: "(%s:%u, %s)" , 1
QString::arg: Argument missing: (%s:%u, %s), 
Critical: Uncaught ReferenceError: THREE is not defined(%s:%u, %s)
```

Updating the `dependencies.txt` file was not solving the issue, but maybe updating `docs/js/viewer.js` does it. If not I will inspect the issue after #435 is merged.

fabienrohrer, 2019-05-08:
After updating dependencies.txt as proposed in https://github.com/omichel/webots/pull/435/files, did you run the docs/local_exporter.py script?

stefaniapedrazzi, 2019-05-08:
probably not.. I will check

DavidMansolino, 2019-05-21:
Can we close this issue?

@stefaniapedrazzi do you still have the error ?

stefaniapedrazzi, 2019-05-21:
Running `.local_exporter.py` was not solving the issue.
But I checked again on `develop` and it seems to work.

PR #435 (2019-05-16): Migrate robot components to THREEjs
Fix #433

- [x] Use the new `webots.min.js` viewer
- [x] Restore the functionalities
    - [x] sliders
    - [x] leds
    - [x] device origins
    - [x] Robot size estimation to set the device origin size
    - [x] motor reset
    - [x] view reset
- [x] Increase rendering quality using textured background and pbr
- [x] Regenerate the robot components (required for some new features like PBR and USE-DEF)
- [x] Side improvements
    - [x] Viewpoint fieldOfView seems wrong
    - [x] Fix robot components on the develop branch (update dependencies.txt)
    - [x]  [... cut]
closed 2019-05-21 by stefaniapedrazzi