cyberbotics/webots #453: --minimize option not working on Ubuntu

fabienrohrer, 2019-05-15:
Launching webots with the `--minimize` option is not working on Ubuntu.

[image]

DavidMansolino, 2019-05-15:
I can't reproduce the problem on Ubuntu 16.04 on master.

DavidMansolino, 2019-05-15:
For me (on Ubuntu 16.04) it is working fine on revision and develop too.

stefaniapedrazzi, 2019-06-13:
Calling `showNormal()` followed by `setWindowState(Qt::WindowMinimized)` seems to work on my Ubuntu 18.04 and the application is not displayed on the screen.

stefaniapedrazzi, 2019-06-13:
I could reproduce the same issue based on a Qt example and I reported it:
https://bugreports.qt.io/browse/QTBUG-76354

closed 2019-06-13 by stefaniapedrazzi

stefaniapedrazzi, 2020-05-04:
I just checked and this Qt bug has been fixed in Qt 5.14.2.
The current version used on Linux is 5.13.1. When updating to the latest Qt version we can get rid of the hotfix introduced in  #563.