cyberbotics/webots #471: three.js webots.min.js loading information

fabienrohrer, 2019-05-20:
The new webots.min.js may take long time to load models. Especially for textures.

Some information about loading time (the lady bug, or a progress bar, or both?) would be welcome.

Could the texture loading be asynchronous as before?

stefaniapedrazzi, 2019-05-20:
> Could the texture loading be asynchronous as before?

Not sure to understand what do you mean.
The texture loading is already asynchronous in `webots.min.js`.
But probably the `render` function is called only after the scene is fully loaded.

fabienrohrer, 2019-05-20:
You're right, we are already close to the optimal: the render function is called once after the x3d load, and once after each texture load. The result is a bit weird but it's difficult to do better.

fabienrohrer, 2019-06-12:
The result after #508 is more comprehensive. The GUI is not blocked during the texture load. I think we can safely close this issue.

closed 2019-06-12 by fabienrohrer