cyberbotics/webots #482: Webots "country" warning

nnuk, 2019-05-21:
I am using Webots on MAC OS. I wanted to convert the OSM file of place in germany to webots file. When I convert it in the terminal it creates a webots file but gives a warning "failed to determine the country" and then it gives this info.

* OSM filed parsed
 * 0 roads generated
 * 0 crossroads generated
 * 0 buildings generated
 * 0 trees generated
 * 0 barriers generated
 * 0 rivers generated
 * 0 areas (forest, water, farmland, etc.) generated
 * 0 parking lines generated

due to this the .wbt has nothing in it but just a plane view. Can you kindly help me wit it?
The OSM file has streets and nodes.

DavidMansolino, 2019-05-21:
Please post the OSM file in this issue.

nnuk, 2019-05-21:
This is the OSM file.
[TESTWEBOTSMAP copy.txt](https://github.com/omichel/webots/files/3203378/TESTWEBOTSMAP.copy.txt)

DavidMansolino, 2019-05-22:
Which version of Webots and which version of python are you using?
I just tried with the latest version of Webots and for me it is working fine.
There is indeed the `failed to determine the country` warning but you can ignore it for now (it is used simply to determine the maximum allowed speed of each road).

But as you can see objects are correctly imported:
```
Warning: Failed to determine the country.
Warning: The exported Road.speedLimit may be wrong because the maxspeed rules is not yet implemented for this country.
  Supported countries:
  - France
  - Germany(default)
  - Japan
  - Switzerland
  - United Kingdom
  - United States
Please contact support@cyberbotics.com to support your country.
 * OSM filed parsed
 * 64 roads generated
 * 35 crossroads generated
 * 3 buildings generated
 * 0 trees generated
 * 0 barriers generated
 * 0 rivers generated
 * 3 areas (forest, water, farmland, etc.) generated
 * 0 parking lines generated
 * map centered with this offset: 470579.850018,5937275.12435).
 * reference coordinates: 53.5835,8.5556.
 * projection used: '+proj=utm +north +zone=32 +lon_0=8.555600 +lat_0=53.583500 +x_0=0 +y_0=0 +ellps=WGS84 +units=m +no_defs'.
Done
```

Here is the resulting world file: [TESTWEBOTSMAP.wbt.zip](https://github.com/omichel/webots/files/3206009/TESTWEBOTSMAP.wbt.zip)

nnuk, 2019-05-22:
For Webots i am using R2018b version and for python it's 2.7.16.

DavidMansolino, 2019-05-22:
Can you please try with the latest version: https://github.com/omichel/webots/releases/tag/R2019a-rev1
We fixed various issues related to the OSM importer since Webots R2018b.

nnuk, 2019-05-24:
Thanks alot for your suggestion. i downloaded R2019a.rev1 and it is now working. The only thing is that it very slow when i try to maximize my map it is really slow, even i tried opening it with my external SSD but still it is slow.

DavidMansolino, 2019-05-24:
Good news!
What is slow, the conversion OSM => Webots, or to open the generated world in Webots?

nnuk, 2019-05-24:
when the map appears in the webots window and i want to enlarge the map to see the roads closely then it takes time to maximize.

DavidMansolino, 2019-05-24:
How big is the map (i.e. how many road/buildings, etc do you have)?

nnuk, 2019-05-24:
I think there 70 odd roads and some crossroads and 16 buildings.

DavidMansolino, 2019-05-24:
Ok than such world should work without any problem in Webots. What is your GPU, are the drivers up to date?

nnuk, 2019-05-24:
The drivers are up to date. My GPU is Intel HD Graphics 5000.

DavidMansolino, 2019-05-24:
Ok then this is probably the problem, we do recommend to have a dedicated AMD or NIVIDIA GPU to be able to run such simulations smoothly.
However, you can try reducing the OpenGL features in the preferences to make it more smooth: https://www.cyberbotics.com/doc/guide/preferences#opengl

nnuk, 2019-05-24:
Thanks alot.

closed 2019-05-24 by nnuk

DavidMansolino, 2019-05-24:
You're welcome.