cyberbotics/webots #483: Background Improvement

DavidMansolino, 2019-05-22:
**Is your feature request related to a problem? Please describe.**
Currently, every simulation requires a Background node with textures to have a nice illumination and reflection of all the objects in the scene.
But this is sometimes a big constraint, for example:
  - To see visually the background is sometimes strange, for example, if the simulation is a room of an apartment, when zooming out it is strange to see the room floating in the environment. In such case, a 'black' visual background would be nicer.
  - As in some simulation, the background is the main source of illumination, it is difficult to vary the light condition, for example when a robot switches off the light. It is possible to change the  `IBLStrength` field of the PBRAppreance node, but this is not possible in a complex scene to change this field for EVERY appearance in the scene.

**Describe the solution you'd like**
I would propose to add two new fields to the background node:
  - SFBool visible: if set to false the background texture is used for illumination/reflection but is not visible (the background color is used).
  - SFFloat  IBLStrength: specifies a scale factor to be applied to the light contribution from the Background.

DavidMansolino, 2019-05-22:
@all This is for now just a proposition if you have other related problem with the current BAckground node or other solution to the problem I explained do not hesitate to add them so that we can have a complete overview of what we need before starting the implementation.

omichel, 2019-05-22:
It looks very good to me. However, instead of `IBLStrength`, I would call it `lightIntensity` which is easier to understand for most users.

DavidMansolino, 2019-05-22:
As just discussed with @omichel, an even better name would be `luminosity` so it doesn't make any possible confusion with lights.

DavidMansolino, 2019-05-27:
As we discussed this morning during the meeting, here are the steps we decided to follow:
  - Remove the `environmentMap` field of the PBRAppearance node: https://github.com/omichel/webots/blob/4f07a74a0631952a7b098e99f0ab92878051e256/resources/nodes/PBRAppearance.wrl#L13
  - Extend the background node to add two new fields(name might still change):
    - `specularIrradianceMap`.
    - `diffuseIrradianceMap` (still need to check, mabe we can generate this map from the other one).
  - Rename the `cubemap` field of the Background node into `skyColorMap`.
  - Cleanup the WbCubmap class to remove all the backing and use instead directly the texture from the two new fields.

DavidMansolino, 2019-05-28:
Just for the curiosity:
**Specular + diffuse:**
[image]
**Specular only:**
[image]
**Diffuse only:**
[image]
**nonde (in that case the background color is used):**
[image]

DavidMansolino, 2019-10-18:
Linked to #880

fabienrohrer, 2019-11-12:
I think this is obsolete. The new background system solves the 2 original issues reported here.

closed 2019-11-12 by fabienrohrer