cyberbotics/webots #488: The PRB appearance images are duplicated many times in the documentation

omichel, 2019-05-23:
Because robot models are using more and more the new PBR appearances, the corresponding JPG images get copied many times in the documentation because they are used by the robot web component. Instead of duplicating these files, we should refer to the original ones to avoid a huge redundancy in our repository.

fabienrohrer, 2019-05-23:
I'm not sure to understand what PBR appearance images are. Do you speak about the textures of the robot component?

omichel, 2019-05-23:
Yes.

fabienrohrer, 2019-05-23:
The situation is still quite tolerable (only few textures are concerned) but it may indeed explode soon especially if we generalize the robot component to objects.

I can do this as I already implemented a similar mechanism to avoid the background textures to be duplicated:
https://github.com/omichel/webots/blob/develop/scripts/web_component_studio/controllers/web_component_studio_supervisor/web_component_studio_supervisor.py#L69
This has to be generalized.

omichel, 2020-06-29:
#1638 should resolve this problem.

BenjaminDeleze, 2021-06-07:
Will be fixed by #2787

closed 2021-07-05 by BenjaminDeleze