cyberbotics/webots #493: wb_supervisor_field_import_mf_node_from_string is slow

schra, 2019-05-23:
**Describe the Bug**

I want to create several thousand objects at run time. For this I use `wb_supervisor_field_import_mf_node_from_string`. Contrary to my expectation, I found that the function is slow.

**Steps to Reproduce**

MWE:

```c++
#include <array>
#include <chrono>
#include <iostream>
#include <webots/robot.h>
#include <webots/supervisor.h>

WbNodeRef root_node;
WbFieldRef root_children_field;

int const insert_position = 0;

void remove_n_objects(int n) {
  for (; n >= 0; --n) {
    wb_supervisor_field_remove_mf(root_children_field, insert_position);
  }
}

void import_n_objects(int n, char const *object) {
  for (; n >= 0; --n) {
    wb_supervisor_field_import_mf_node_from_string(root_children_field,
                                                   insert_position, object);
  }
}

int main() {
  wb_robot_init();

  root_node = wb_supervisor_node_get_root();
  root_children_field = wb_supervisor_node_get_field(root_node, "children");

  std::array<int, 15> num_objects_to_measure{{100, 200, 300, 400, 500, 600, 700,
                                              800, 900, 1000, 1500, 2000, 2500,
                                              3000, 3500}};

  for (int n : num_objects_to_measure) {
    auto start = std::chrono::system_clock::now();
    import_n_objects(n, "Group { children [] }");

    auto diff = std::chrono::duration_cast<std::chrono::milliseconds>(
        std::chrono::system_clock::now() - start);
    std::cout << n << "," << diff.count() << std::endl;
    wb_robot_step(1); // show std::cout

    remove_n_objects(n);
  }

  wb_robot_cleanup();
  return 0;
}
```

Note 1: The cache is warm in this code. So the actual times are even higher (you can try this, if you restart Webots for each measurement)

Note 2: I use `Group { children []  }` here in order to exclude e.g. overhead that is associated with loading more complex objects (e.g. when textures have to be loaded).

Note 3: If you duplicate the string `n` times into one string and then call `wb_supervisor_field_remove_mf` with this string, you will get similar measurements.

On my machine I get the following data (in CSV format):

```
num_objects,duration
100,259
200,565
300,1053
400,1539
500,2259
600,2942
700,3895
800,5138
900,6152
1000,7480
1500,14800
2000,22322
2500,28397
3000,35531
3500,43432
```

I created a plot of this data. You can see that in the start the run time is quadratic and then it's almost linear:

[image]

**Expected behavior**
 I expected it to take at most one second for a couple of thousand objects.

DavidMansolino, 2019-05-24:
Instead of importing the nodes one by one you could try the following:

1. Create a string with all the node and call wb_supervisor_field_import_mf_node_from_string only once (e.g. `wb_supervisor_field_import_mf_node_from_string(root_children_field, insert_position, "Group { children [] } Group { children [] } Group { children [] }");`)
2. From your supervisor generate a world file with all the required nodes and then load this file using `wb_supervisor_world_load` (https://www.cyberbotics.com/doc/reference/supervisor#wb_supervisor_world_load).

schra, 2019-05-27:
Hey David,

your first method is significantly faster than my method but it is also not really fast. I measured the following durations:
* 2.7 seconds for 3000 `Group` objects
* 3.9 seconds for 3500 `Group` objects

Maybe it wasn't a good idea to demonstrate my problem with `Group` objects. I adjusted my MWE so it makes use of `Solid` objects and implements both my method and your first method:

```C++
#include <array>
#include <chrono>
#include <iostream>
#include <string>
#include <webots/robot.h>
#include <webots/supervisor.h>

WbNodeRef root_node;
WbFieldRef root_children_field;

int const insert_position = 0;

void remove_n_objects(int n) {
  for (; n >= 0; --n) {
    wb_supervisor_field_remove_mf(root_children_field, insert_position);
  }
}

void import_n_objects_once(int n, char const *object) {
  std::string objects;
  for (; n >= 0; --n) {
    objects += std::string(object);
  }
  wb_supervisor_field_import_mf_node_from_string(
      root_children_field, insert_position, objects.c_str());
}
void import_n_objects_individually(int n, char const *object) {
  for (; n >= 0; --n) {
    wb_supervisor_field_import_mf_node_from_string(root_children_field,
                                                   insert_position, object);
  }
}

int main() {
  wb_robot_init();

  root_node = wb_supervisor_node_get_root();
  root_children_field = wb_supervisor_node_get_field(root_node, "children");

  std::array<int, 12> num_objects_to_measure{
      {100, 200, 300, 400, 500, 600, 700, 1500, 2000, 2500, 3000, 3500}};

  for (int n : num_objects_to_measure) {
    auto start = std::chrono::system_clock::now();
    import_n_objects_once(n, "Solid { }");

    auto diff = std::chrono::duration_cast<std::chrono::milliseconds>(
        std::chrono::system_clock::now() - start);
    std::cout << n << "," << diff.count() << std::endl;
    wb_robot_step(1); // show std::cout

    remove_n_objects(n);
  }

  wb_robot_cleanup();
  return 0;
}
```

It yields the following measurements:

[image]

Note that your second method is even slower: I tried to open a world with 3500 `Solid` objects and it took longer than 3 minutes to load (I just stopped after 3 minutes).

DavidMansolino, 2019-05-28:
I am sorry but I just tried and I can open a world with ~3000 Solid nodes in ~2 seconds on my computer.
Can you please try opening this world and let me know how long it takes?
[empty.wbt.zip](https://github.com/omichel/webots/files/3226216/empty.wbt.zip)
Can you also give us the output of the `webots   --sysinfo` command just to make sure that we are comparing with a ~similar setup.

schra, 2019-05-28:
> I am sorry but I just tried and I can open a world with ~3000 Solid nodes in ~2 seconds on my computer.
> Can you please try opening this world and let me know how long it takes?

I tried your world and you are correct. In my previous post I accidentally added my program from above as a controller. Your world also loads quite fast for me.

Unfortunately, the project I'm working on prohibits me from using `wb_supervisor_world_load` since I have to create new objects at runtime (`wb_supervisor_world_load` would stop the simulation).

Do you consider `wb_supervisor_field_import_mf_node_from_string` being slow a bug or not? For me it's critical that `wb_supervisor_field_import_mf_node_from_string` only takes a couple of seconds maximum for several thousand objects (and I've shown above that this currently takes a lot more than that).

> Can you also give us the output of the webots --sysinfo command just to make sure that we are comparing with a ~similar setup.

```
$ webots --sysinfo
System: Linux 4.15.0-50-generic x86_64
Processor: Intel(R) Xeon(R) CPU E3-1505M v6 @ 3.00GHz
Number of cores: 8
OpenGL vendor: NVIDIA Corporation (0x8086)
OpenGL renderer: Quadro M620/PCIe/SSE2 (0x591d)
```

DavidMansolino, 2019-05-28:
Good to know that we have similar results for loading this world (otherwise this would definitely be a bug).

Ok, your computer setup seems correct (even if the last line is missing from your message).

> Do you consider wb_supervisor_field_import_mf_node_from_string being slow a bug or not? For me it's critical that wb_supervisor_field_import_mf_node_from_string only takes a couple of seconds maximum for several thousand objects (and I've shown above that this currently takes a lot more than that).

The version in which you import all the nodes at once should indeed be faster (it should be similar to the world with already 3000 solids). I will make some tests an let you know if I can find something.

Alternatively, isn't it possible for you to have already the objects in the world and use always the same one and simply 'resetting' them?

DavidMansolino, 2019-05-28:
I have made some tests and this slow-down that you can see when importing a lot of nodes is because at each node import the dictionary (which is managing the USE-DEF) is updated (which is becoming costly when it has to parse many nodes already present in the scene):
https://github.com/omichel/webots/blob/9c3fa83be3191221a2d2c38f5995339c5c692e3d/src/webots/nodes/utils/WbNodeOperations.cpp#L322
(removing this call I can import 5000nodes in ~30sec).
There is for sure room for optimization (e.g. updating the dictionnary only if the new node contains DEF-USE). Unfortunately, this is not on our top-priority roadmap. But if you want to have a look we would be glad to guide you.

DavidMansolino, 2019-07-16:
Unfortunately optimizing this is not on our short term road map as some alternatives exist.

closed 2019-07-16 by DavidMansolino

schra, 2019-07-19:
> I have made some tests and this slow-down that you can see when importing a lot of nodes is because at each node import the dictionary (which is managing the USE-DEF) is updated (which is becoming costly when it has to parse many nodes already present in the scene):

Thanks providing this mitigation! It certainly reduces the loading quite a bit (but it's still slow)

> Unfortunately optimizing this is not on our short term road map

I understand that you do not plan to fix this issue.

>  as some alternatives exist.

What alternatives are your speaking of? If I understood you correctly the two alternatives are `wb_supervisor_world_load` and `wb_supervisor_field_import_mf_node_from_string`. As I stated above I can't use the former and the latter is too slow (both calling it once or n times). Do you have another alternative in mind?

DavidMansolino, 2019-07-22:
Hi, yes indeed I was speaking about `wb_supervisor_world_load` and `wb_supervisor_field_import_mf_node_from_string`, I understand they might not work out of the box in your case and might require to adapt a bit your setup.