cyberbotics/webots #502: Hyperlinks no longer visible after clicking on it

stefaniapedrazzi, 2019-05-27:
**Documentation page(s) concerned**
Any page containing an hyperlink

**What is the problem with the current documentation?**
Hyperlink text color is blue and they are visible in the text.
But after clicking on it, the color is changed to black and they are no longer visible.

**Describe how you would fix it**
Improve the color definition in the CSS.

fabienrohrer, 2019-05-28:
Just to be sure we agree on the description of this issue: this seems to occur only on hyperlink targeting an external link, and only once per page, right?
Internal links seems to behave correctly.

stefaniapedrazzi, 2019-05-28:
No, I get this issue with internal links: for example when in the reference manual with link to other node pages.
Note that it is not only the visited rule that changes the color to black but also the hover rule.

stefaniapedrazzi, 2019-05-28:
But after you click on the page and unselect the link, then it returns to the original link (non-visited) color, because it is the "active" rule that sets the text color to black.

fabienrohrer, 2019-05-28:
Ok. It seems that I can only reproduce the "hover" issue on the macOS browsers, and the "active" issue seems to only occur on external links in my environment.

fabienrohrer, 2019-05-28:
Fixed in #504

PR #504 (2019-05-28): Improve hyperlink colors (hotfix)
Fix #502 

Process: start from the [original hyperlink color](https://www.colorhexa.com/317589), increase luminosity (along the monochromatic scale) for the unvisited hyperlinks, decrease the luminosity for the hyperlink events (active / hover). The difference is subtle but visible, good looking IMO.

Preview: https://cyberbotics.com/doc/guide/samples-environments?version=hotfix-hyperlink-colors
closed 2019-05-28 by fabienrohrer