cyberbotics/webots #534: Continuous Integration Strategy

DavidMansolino, 2019-06-05:
To summarize our discussion with @omichel  in the office here is the strategy we want to adopt for the continuous integration test.
The tests will be done in several independent stages (a new VM is used for each of them).
Here is the list of stages (each stage imply the previous one nd is launched only if the previous one sucedded):

## Sources Tests
This test should be very fast (~5min max)
- At **each commit** on a non-draft PR / manually triggerd.
- On all platforms / or linux only if this slows done tests.
- Contains the following tests:
    - doc/tests/...
    - tests/sources/test_clang_format (as this test is slow we will run it only on the modified files)
    - tests/sources/test_cppcheck (as this test is slow we will run it only on the modified files)
    - tests/sources/test_icons
    - tests/sources/test_license
    - tests/sources/test_matlab_functions
    - tests/sources/test_pep8
    - tests/sources/test_project_structure
    - tests/sources/test_proto_names
    - tests/sources/test_textures
- In case of daily test (CRON) we test the maximum of OS + setup and the tests are made on all the files (not only the modified one)

## Unit Tests
This test is quite long  (> 1h)
- This test **runs daily** / manually triggerd / triggerd by the `travis unit` keyword in the commit message.
- On all platforms + all setup
- Contains the following tests (webots beeing compiled in debug mode):
    - test_suite
    - ros tests (at this point we will be able to **drop Jenkins**)
    - tests/manual_tests/test_worlds_warnings.py
    - src/packaging/update_all_worlds.py

## Distribution
This test is quite long  (> 1h)
- This test **runs daily** / manually triggerd / triggerd by the `travis distrib` keyword in the commit message.
- On all platforms
- Contains the following tests (webots beeing compiled in distrib mode):
    - create the distribution.
    - save the distribution to a temporary folder (e.g. google cloud)

## Deploy
This test is only to deploy files on github
- This test **runs daily** / manually triggerd / triggerd by the `travis deploy` keyword in the commit message.
- On all platforms
- This test get all the distributions from the temporary folder and upload them as a github pre-release.

DavidMansolino, 2019-06-20:
The `tests/manual_tests/test_worlds_warnings.py` and `src/packaging/update_all_worlds.py` tests are going to be difficult to run on Travis as they are very slow to run on Travis (more than 1h just for the test for  `src/packaging/update_all_worlds.py`).

omichel, 2019-06-21:
I believe these tests should be run on our machines prior to a release. We should update the release procedure to add these tests.

omichel, 2019-06-21:
I just moved the release procedure to the wiki of this repository and updated it accordingly.

DavidMansolino, 2019-06-21:
Thank you, I just also updated some outdated parts (e.g. references to webots-doc repo).

DavidMansolino, 2019-07-16:
I believe we can close this issue as most of it is now functional and we decided to keep the rest (e.g. `src/packaging/update_all_worlds.py`) out of the CI.
Furthermore, this is now documented on the wiki: https://github.com/omichel/webots/wiki/Continuous-Integration

closed 2019-07-16 by DavidMansolino