cyberbotics/webots #582: Rounding issues saving float values to WBT file

stefaniapedrazzi, 2019-06-18:
There are rounding issues writing float values to world file.
This only occurs for some specific values, for example 1.6 but not 1.5.

**Steps to Reproduce**
1. Open new world
2. Change `Viewpoint.far` to 1.6
3. Save the world
-> in the WBT file the far value is exported as `far 1.5999999999999998`

**Expected behavior**
The export field value should be `far 1.6` and not `far 1.5999999999999998`

**System**
 - Operating System: Linux Ubuntu 18.04

stefaniapedrazzi, 2019-06-18:
The value is already rounded in Webots before writing it to file.

stefaniapedrazzi, 2019-06-18:
On my PC the correct value `1.6` is changed to `1.599999999999999` here: https://github.com/omichel/webots/blob/eafc9e5324690b1250696d790df982a788425000/src/webots/maths/WbPrecision.cpp#L74

stefaniapedrazzi, 2019-06-19:
The problem comes from the fact that when editing from the field editor, the internal field value in Webots is stored with a precision (GUI_MEDIUM) lower than the one used when writing the value to file (DOUBLE_MAX).

Fixing the way the values are rounded in the field editor code also fixes the problem on WBT write:
* store internal value with DOUBLE_MAX precision
* round value to GUI_MEDIUM when displaying it to in the field editor

stefaniapedrazzi, 2019-06-20:
Actually it doesn't seem to fix the problem: for some values it works but not for others.
The value set by the QDoubleSpinBox often it is not a rounded value: for example `0.29999999999999998890` when typing 0.3.
And sometimes the `WbPrecision::roundValue` function with DOUBLE_MAX precision in some cases increases the distance from the rounded value: for example from `0.29999999999999998890` to `0.30000000000000004441`.

omichel, 2019-06-20:
This is inherent to the nature of IEEE floating point representation, as explained here: https://0.30000000000000004.com.

stefaniapedrazzi, 2019-06-20:
Yes. But if we want to set and store rounded values (with a high precision) we cannot loose precision in between.

stefaniapedrazzi, 2019-06-20:
Removing this rounding:
https://github.com/omichel/webots/blob/eafc9e5324690b1250696d790df982a788425000/src/webots/scene_tree/WbDoubleEditor.cpp#L84
makes the storing work correctly much more often than even increasing the precision to DOUBLE_MAX. But still sometimes it fails.

stefaniapedrazzi, 2019-06-20:
Maybe I found a working method.

DavidMansolino, 2019-06-26:
It seems the issue is not fixed. In #656 I have tons of rounding issues.
To reproduce the problem, simply open the battery.wbt world, move one of the block with the arrows and save:
```diff
--- a/projects/samples/devices/worlds/battery.wbt
+++ b/projects/samples/devices/worlds/battery.wbt
@@ -62,7 +62,7 @@ DEF BLUE_BOX Solid {
   }
 }
 DEF YELLOW_BOX Solid {
-  translation -0.2 0.05 0.15
+  translation -0.12631299999999998 0.049999999999999996 0.118848
   rotation 0 1 0 0.4
   children [
     Shape {

```

DavidMansolino, 2019-06-26:
A probably linked issue is that if you change the `Solid.rotationStep` field, then it is not possible to restore the default value, the default value is restored but still appears as non-default (probably because of the rounding).

stefaniapedrazzi, 2019-06-27:
Just for note:
- the issue with the default value seems to be introduced in #627 
- issues when moving objects with the handles was already reproducible before #627

stefaniapedrazzi, 2019-07-01:
Removing the rounding here fixes the problem with the handles:
 https://github.com/omichel/webots/blob/3a446d433e81c5cb293c593316438631a0d6b49e/src/webots/gui/WbDragTransformEvent.cpp#L91

stefaniapedrazzi, 2019-07-01:
It seems that reverting the rounding introduced in #266 to fix rounding issue solves the problem at least on my system..

PR #627 (2019-06-20): Hotfix rounding fields values on WBT write
Fix #582: it seems that using `QDoubleSpinBox::text().toDouble()` instead of `QDoubleSpinBox::value()` solves the issue.
PR #684 (2019-07-01): Hotfix rounding issues
Fix #582: improve fix in #627 by changing the rounding method so that it uses the `WbPrecision::doubleToString` rounded value.

Using the string value to round reduces differences when saving the value to file and make it more robust to the intrinsic floating noise.
Note that for almost all the values the two rounding methods are equivalent.

This is the same principle it was used in #627 where the value was extracted from the spin box text, but now it has a higher precision that fixes the reset to default issue.
closed 2019-07-01 by stefaniapedrazzi