cyberbotics/webots #623: Wrong rendering after loading many worlds

stefaniapedrazzi, 2019-06-20:
After loading many worlds, at some time the rendering becomes much darker than what it should do.
For example starting Webots and running the Guided Tour, at some point I got this appearance for the `receiver_noise.wbt` simulation:

[image]

Then, I quitted and restarted Webots and the rendering was correct:

[image]

I could reproduce this behaviour a couple of times (at some point the also the hokuyo.wbt Display overlays were not shown any longer on the scene).

Probably my NVIDIA drivers are not up-to-date, but it would be strange if this is the cause of the issue because it works at the beginning.

System: Windows 10
Graphics card: NVIDIA GeForce GT 640M with 2GB VRAM
OpenGL version: 3.3.0 NVIDIA 418.81
Ambient Occlusion: Medium
Texture Quality: High

omichel, 2019-06-20:
I can't reproduce this on my system:
```
$ webots --sysinfo
System: Windows 10 64-bit
Processor: Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz
Number of cores: 4
OpenGL vendor: NVIDIA Corporation (0x10de)
OpenGL renderer: GeForce GTX 970/PCIe/SSE2 (0x13c2)
OpenGL version: 4.6.0 NVIDIA 388.13
```
However, I noticed twice a random crash when running the `animated_skin.wbt` simulation (5th simulation of the guided tour). Can you reproduce the problem you experienced if you don't load the `animated_skin.wbt` before? If so, I would suspect a corruption related to this simulation...

stefaniapedrazzi, 2019-06-20:
Yes, I could reproduce it just running the devices worlds.

omichel, 2019-06-20:
And can you reproduce it by simply reverting many times `receiver_noise.wbt`?

stefaniapedrazzi, 2019-06-20:
I will check.
When I took the screenshots I started from the gps.wbt device worlds (because I just had the same issue with hokuyo.wbt)
I'm sure that it the problem is not in the `receiver_noise.wbt` world, because if I opened other worlds after it I had the same effect. But it would of course be good to know if there is a specific world causing issues and `receiver_noise.wbt` is simple enough to exclude basic issues.

stefaniapedrazzi, 2019-06-20:
I tried but I cannot reproduce it, neither reverting the `receiver_noise.wbt` nor running other devices worlds.

stefaniapedrazzi, 2019-11-28:
I will close this issue as I could not reproduce it since a while.

stefaniapedrazzi, 2019-12-17:
I could reproduce it again when testing Webots R2020a in a VM (low texture quality, disabled shadows and disabled ambient occlusion and disabled anti-aliasing).
I was running the devices guided tour and the problem occurred again when loading the `receiver_noise.wbt` (after `range_finder.wbt`).
In the console I got many errors:
```
OpenGL error: GL_INVALID_OPERATION
```

I tried to reload or switch to other worlds, but the rendering was only fixed after restarting Webots.

stefaniapedrazzi, 2019-12-17:
I can finally reproduce this error systematically:
1. run `range_finder.wbt`
2. double click on the overlay to open the range-finder image in a separate window
3. open `receiver_noise.wbt`

stefaniapedrazzi, 2019-12-17:
I checked and I can reproduce it also on my development environment.

Related to #882 and #302.

PR #2119 (2020-08-19): Fix issues with extern rendering device windows
Tasks:
  - [x] fix #623 and #882:
    when deleting the rendering device window the OpenGL context was not the correct one
  - [x] fix  #302:
    if the rendering device window is open before the device is setup, the initialization of the textures, VAO and VBO is skipped and the image is not displayed
  - [x] fix the code so that it really uses VAO (currently just declared but not used)
  - [x] update change log
  - [x]  update test_suite mirror test and add regression tests for #882 and #302
closed 2020-08-19 by stefaniapedrazzi